Ralim/IronOS #306: OLED Screen flickers as going to sleep

dine909, 2018-05-22:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ X] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
report a bug

* **What is the current behavior?**
Operation is normal, if left for sleep timeout, OLED flickers off fast at first slowing down then eventually remains off, when picked up the OLED comes back on as normal.

* **What is the expected behavior?**
OLED should go off and stay off

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. Leave stationary until timeout
2. 
3. 

***Video of problem if hard to reproduce***
On request

* **What is the motivation / use case for changing the behavior?**
It looks wrong

* **What are you running:**
TS100  V2.05 PCB2 MAY 7 2018

  - Firmware Version: 2.05
  - PCB Version: 2
  - Power Supply 18V lithium pack (5s) :

TheGiolly, 2018-05-27:
I have the same issue as you

dine909, 2019-02-27:
Can the temp be averaged and that used as a trigger instead?

Ralim, 2020-09-14:
Is this issue still occuring for anyone? 😰

closed 2020-11-01 by Ralim