Ralim/IronOS #309: PID is pulsing when cooling down from boost, at unreasonable delta temperature.

dine909, 2018-05-25:
* **I'm submitting a ...**
  - [X] Bug report
  - [] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behaviour?**
PID is pulsing when cooling down from boost, which in turn takes ages to get back to normal temperature.

* **What is the expected behaviour?**
I would expect the PID output to be masked until the over-temperature value is within a sane controllable range say - 15'C (this is a commonly done in PID systems to allow fine tuning and responsivness)

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. Heat iron up to a temperature
2. Apply boost >100'C of normal setting
3. Release boost
4. Observe the element immediately being pulsed even though current temperature is 100'C off the target

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**
To speed up the time taken from boost to get back normal temperature.

What are you running:

Firmware Version: 2.05
PCB Version: 2
Power Supply 18V lithium pack (5s) :

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.x
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :

Video of problem if hard to reproduce
On request

blitzcode, 2018-05-25:
I also see this issue. Related, the TS100 never goes quite down to its sleep temperature. It's always heating a bit so it never cools to 150C/120C or whatever the default was.

dine909, 2018-05-25:
Yes, i see this behaviour too - i felt it’s the same issue to i didn’t include it explicitly as not to confuse things.

> 
> I also see this issue. Related, the TS100 never goes quite down to its sleep temperature. It's always heating a bit so it never cools to 150C/120C or whatever the default was.

-- 

Showave Ltd.

t:	02081238404

w:	showave.co.uk <http://showave.co.uk>

Registered in Scotland, UK No. SC426945

This e-mail and any attachments 
may contain privileged and/or confidential information. If you are not the 
intended recipient, you are hereby notified that any dissemination, 
distribution or copying of the information contained in this message is 
prohibited. If you have received this message in error, please notify the 
sender by reply e-mail and delete the message from your system. Any 
statements in this message which do not relate to the business of Showave 
Ltd., are neither given nor endorsed by Showave Ltd.

closed 2018-07-27 by Ralim