Ralim/IronOS #312: Cant update DFU 3.43( my pc wont recognize)

michael-blue2, 2018-06-03:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ ] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**

* **What is the current behavior?**

* **What is the expected behavior?**

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. 
2. 
3. 

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.x
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :

* **Other information** 

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Firebie, 2018-06-04:
Try another USB cable and/or another USB port

Ralim, 2018-07-27:
Closing due to inactivity.

closed 2018-07-27 by Ralim

Repled, 2018-07-28:
Is it possible to comment here....

Repled, 2018-07-28:
My experience is that it is Windows 10 that is almost not working at all. Had to try 50-100 times to make it bite and I never got any .RDY at the end. Going to an old Win 7 computer and it worked like a charm at once.