Ralim/IronOS #313: Save and exit not working

rakusan2, 2018-06-06:
* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **What is the current behavior?**

Upon calibrating the voltage I have tried holding down both buttons to exit and nothing happened.
It did not freeze, it just did not exit

* **What is the expected behavior?**
holding down both buttons exits the input voltage setting

***Steps to reproduce:***

1. Go to calibrate voltage in the advanced menu
2. Hold both buttons

* **What is the motivation / use case for changing the behavior?**
To be able to get out of the menu

* **What are you running:**
  - Firmware Version: 2.04.1
  - PCB Version: (2)
  - Power Supply (Voltage and Current Rating) : 12.08V, 2A

rbzikadze, 2018-06-15:
Can you try holding back button after setting the voltage and see if it exits the menu? I agree it's a bug and it should exit when both buttons are pressed, it's an easy fix tho

simontom, 2018-06-21:
Hi, yes, if I press Button B for like 3 secs it gets me out of the menu.

rakusan2, 2018-06-21:
Holding the B button has also gotten me out of the menu, tho I would recommend changing the README to reflect this

Ralim, 2018-07-27:
Holding either button, or the pressing both combination should exit the menu. The double press is more timing sensitive than the others however.

closed 2018-07-29 by Ralim