Ralim/IronOS #314: Feature - Fall detect

MR-Zon, 2018-06-11:
Great Firmware Thank you!

* **I'm submitting a ...**
  Feature request

* **Do you want to request a *feature* or report a *bug*?**
 Feature.
Would it be possible to detect a free-fall (off desk), and possible impact and switch off? 
acceleration of over 9m/s2 for 0.2sec-0.3sec.
I do not know what 'normal use' gives in acceleration. this might make the feature non viable.
The sleep function does reduce the need for this feature. 

* **What is the motivation / use case for changing the behavior?**
Motivation : I have burnt holes in carpets before (not with this iron).
This would be a safety feature.

Thanks again for your work.

LarsSimonsen, 2018-06-22:
Impact detection would be easiest to implement, and then you could turn it off by giving it a whack. Nice idea. 

Not sure if it would save your carpets, though. It does take some time to cool down.

deadbird99, 2018-07-10:
Very nice idea, indeed! Looking forward to its implementation!

dhiltonp, 2018-09-06:
It looks like MMA8652FC has free-fall detection built in, which is pretty handy!

> The detection of “Freefall” involves the monitoring of the X, Y, and Z axes for the condition where the acceleration magnitude is below a user-specified threshold for a user-definable amount of time. Usable threshold levels are typically between ±100 mg and
±500 mg.

Ralim, 2021-09-07:
Yeah this is absolutely a duplicate and its never really been done as its benefit is quite small compared to both the complexity of detection and the tuning required to avoid false positives.