Ralim/IronOS #317: v2.04.1 No Cyrillic

Ashdinalt, 2018-06-17:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ *] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Bug report

* **What is the current behavior?**
It does not show any words in Russian or Ukrainian. Just blank spots, where the words supposed to be.
In 2.04RC2 there were no issues with Cyrillic fonts

* **What is the expected behavior?**

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. Flash 2.04.1 RU or UA
2. 
3. 

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.x
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :

* **Other information** 

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

vnishukov, 2018-07-20:
Have the same issue. Any updates of this issue?

Ashdinalt, 2018-07-29:
Nope. Ralim kinda ignores this issue. I use the older version now.
v2.05 released, but no Russian version there.

Ralim, 2018-07-29:
It's not that I ignore this issue, its on my to-do list, but unfortunately languages and fonts are usually the biggest pain of this entire project, Cyrillic seems to be the worst of these whenever we try and use link time optimization which is required for Russian to even fit in the firmware. This is why 2.05 doesn't have Russian at the moment, is that it didn't fit and I wanted to push *something* that push nothing for even longer. If you find a fix for this you are always welcome to send a pull request with a fix or workaround at any time. I work on this firmware when I have time, which is very varied and can change rapidly depending on what else happens in my life.

This is made worse by every so often someone commenting in all capitals about Russian, or raising extra issue noise about it not being there, as if I don't know about it. I'm well aware, but as I have limited time I try and tackle fundamental use case issues first and then come to languages once those are sorted. I cant miss a screen full of compiler warnings when the firmware didn't fit, and generally I push up what I can, and then come back to the ones with issues when I have time to work on them.

Ashdinalt, 2018-07-29:
Yes, I understand, that the project is fully enthusiastic, and it's really great that someone is capable of doing it at all. I really appreciate that. And that's why I never updated the issue since June 17. It is just about a reply. Like 1,5 months with no reply. You should have just thrown a line - "Know the issue, working on it". So we just know that we should wait for a fix.

Ralim, 2018-08-02:
Can you confirm if either 2.05 or 2.05.01 still have this issue for you ?

Ashdinalt, 2018-08-02:
Doesn't work at all. Black screen. When pressing button 1 shows DFU: 3.43 and that's it

Ralim, 2018-08-02:
Can you confirm if you mean pressing the front button jumps you to DFU, or if powering up with the button held jumps you to DFU? Pressing the front button shouldnt enter dfu mode unless its rebooting / crashing hmm

Ashdinalt, 2018-08-02:
No, no. After flashing 2.05.01_RU, when powering up, regular powering up, without pressing anything - black screen. Only reaction after that is on pressing front button, which leads to showing "DFU: 3.43".
Flashed 2.05_EN - works fine (in English of course)

bakerenko, 2018-08-03:
same with me. As Ashdinalt wrote. No working firmware since 2.04

vladkorotnev, 2018-08-05:
Same here, 2.05.01 just flashed and indeed RU version isn't working.

vampirio-ru, 2018-08-08:
I confirm, on 2.05.01 russion version firmware does not working. Empty place instead of text.

freetushkan, 2018-08-08:
Same. 2.05.01 doesn't show cyrillic letters.

vampirio-ru, 2018-08-30:
repompiled working version here: https://files.fm/u/4c6rsr2j

vampirio-ru, 2018-08-31:
today i've try to compile under "Ubuntu" and "Gentu" in "Windows Subsystem for Linux" using same compiller as above, and translation works fine.

way to reproduce:
1. need Windows 10
2. make steps from https://docs.microsoft.com/en-us/windows/wsl/install-win10
3. go to Linux console
5. sudo apt update
6. sudo apt upgrade
7. sudo install gcc-arm-none-eabi
8. sudo install make
9. compile using "build.sh" from the project

Ralim, 2018-08-31:
@vampirio-ru 
Those are the exact steps I used to build the 2.05.1 release you see in the releases tab of this repository. :/

vampirio-ru, 2018-08-31:
way to reproduce timers bug

currently i'am using latest hex file from git (EN version) v2.05.01

1. in configuration: sleep
   sleep temp: 090
   sleep timeout: 01M
   shutdown timeout: 03M
   motion sensetivity: 7
2. in configuration advanced:
   detailed idle: on

3. go to solidign mode,
setting working temp in 160,
put solider on my desk and dont touch anything,
wait ~5 min and no reaction from solider

then i made downgrade to v2.05 (EN): problem exist.
then i made downgrade to v2.04.1 (EN):  WAS NO PROBLEM.

closed 2019-08-27 by Ralim