Ralim/IronOS #320: The TS80 (TS200) iron prototype has been showcased.

JohnEdwa, 2018-06-25:
MarcoReps just [released a video showcasing the current prototype of the new TS80.](https://youtu.be/83STpUXHilk?t=48)

TL;DW, so far it has:

* Smaller full metal case.
* Swappable front grip part.
* Different tip design with 3.5mm stereo jack as the connector.
* USB-C with QC 3.0, meaning max power should be 18W - slow to heat up, but performs better than the TS100 @ 24V during testing.
* STM32 F103T8U6 MCU
* PCB with TS200 V1.6 written on it :)

marvedjensen, 2018-06-26:
nice

Firebie, 2018-06-26:
It looks like should be a new case for TS100 (it is shown at the first photo of carrying pouch) with additional protection for fingers.

marvedjensen, 2018-06-26:
But dont Like the USB 3.0 9v just Got the 24v 5525dc power supply for the ts100

dine909, 2018-06-26:
Why are you using the issue tracker for this? that's what reddit is for. 

Please consider deleting this and taking the discussion elsewhere, and leaving the author and subscribers inbox un-bombed.

Many thanks.

JohnEdwa, 2018-06-26:
@dine909 
I'd be glad to take the discussion to the board/subreddit/forum/post/pub/streetcorner, but there isn't one. Which to be honest is rather strange. Know of any forums that would allow Github openID logins we could set it up on?

But as the TS80 is the successor of the TS100 I thought people here would be rather interested in seeing how the prototype is shaping up. If Ralim feels this isn't appropriate, he can remove it.

JennyEverywhere, 2018-06-26:
You could try at https://www.reddit.com/r/ts100/ as a start.

provalinf, 2018-07-04:
I'm not sure if this discussion will be deleted or not, but to come back to the subject, the fact that it is not Power Delivery 2.0 compatible might be problematic, even if the temperature is "super well controlled". 15-18W will not replace 65-100W.
On the other hand, I will from this step preorder the cover for my TS100, thanks for the link :))

marvedjensen, 2018-07-04:
both my ts100 have the bad tip/sens error  problem

CornyjK, 2018-07-09:
Powering it would be more difficult, I will have to implement USB PD @ 20 V, I just wish they kept power jack as option.

marvedjensen, 2018-07-10:
Why Not have it Like on the ts100 with dc Jack and USB right next to the dc Jack just opgrade it to USB c and keep det dc Jack

jj-uk, 2018-07-10:
This thread should be deleted / moved to reddit.

The OP-provided links are affiliate links, which were added after the original feature request was opened.

The OP is clearly using this thread as an attempt to make money.

marvedjensen, 2018-07-10:
HK-uk What Are u talknig about the link to youtube ?

jj-uk, 2018-07-10:
@marvedjensen  - Cookies issue my end. Comment deleted.

jc260552, 2018-07-10:
Surely the final version usb 3 port will support 50w+

Eldenroot, 2018-08-05:
So what will be the max power, because TS100 with 24V is just superior. I dont like USB3 connector, design is nice but I dont think I will sitch to TS80 from TS100... no extra value, maybe worst.

bobafetthotmail, 2018-08-11:
There is a revew of the T80 from EEVblog https://www.youtube.com/watch?v=_Z9es-D9_8g and it looks like it does its job well, it does punch way above what its rated power would suggest.

Ralim, 2018-08-12:
There will *eventually* be TS80 support in this firmware too.
Noting public so far but i do have local support working for the hardware roughly, but need to update support for the tip drive system in this firmware and then figure out temp curve of its tips :)

JennyEverywhere, 2018-08-12:
The TS80 was just reviewed on EEVBlog's Youtube channel.
https://youtu.be/_Z9es-D9_8g

bobafetthotmail, 2018-08-12:
> Here's hoping they will make TS100 versions of the TS80 tips, I'd like a short and stubby chisel tip like that.

The Hakko T12/15 tips are compatible with the TS100, although they are longer.
You will probably need the hakko rubber grip to be able to grip the iron near its new point of balance https://www.amazon.com/Hakko-B3218-Blue-Anti-Bacterial-FM-2027/dp/B0015X6BPY
and a 3D printed adapter to fit firmly the hakko rubber grip on the TS100 (othwerise it spins)
https://www.thingiverse.com/thing:2739632
https://www.thingiverse.com/thing:3021068

joric, 2018-08-20:
@Ralim did minidso publish ts80 firmware yet? Can't find it on their forum.

Ralim, 2018-08-20:
Nothing's public yet, but not worried already started work on a firmware for it. Mostly need to do temp curves for tips and figure out quick charge

joric, 2018-08-25:
@Ralim ok well is there anything non-public? For starters, did you make a firmware dump? I assume you can't program it back via bootloader, so you hooked up to the SWD pins? Is there a bootloader? Is firmware compatible with ts100? This is just so very interesting!

Ralim, 2018-08-26:
@joric
Hardware is very similar to ts100, close enough it's mostly just different pinouts and adding the QC negotiation and it's compatible.

Very similar bootloader, same update method.

Haven't actually dumped the bootloader yet. Would be curious if it could be put in a ts100 if it's more tolerant.

There is the firmware branch on this repo for it now.

Things left to do I'll move to a new issue soon.

closed 2018-10-18 by Ralim