Ralim/IronOS #322: .gitignore references files that are currently tracked

Cabalist, 2018-06-28:
* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **What is the current behavior?**
The following lines are in the .gitignore but there are files in the repo that match these.   Is that intentional?

```
*.cache
workspace/TS100/.settings/language.settings.xml
```

These match:

```
./Bootup Logo/Logo GUI/TS100 Logo Editor/TS100 Logo Editor/obj/Debug/CoreCompileInputs.cache
./workspace/TS100/.settings/language.settings.xml
```

* **What is the expected behavior?**

Entries in the .gitignore are not part of the repo.

I found this while looking through the code and obviously low priority but I thought I'd ask before submitting a PR.

Ralim, 2018-06-29:
Feel free to throw up a PR to drop these files, nothing under the Debug folder should ever be stored in git. Probably my bad ages back at some point :/

Cabalist, 2018-06-29:
No worries!   I just submitted #326 which takes care of these and a few others that have built up.   I updated the .gitignore as well to prevent this in the future.  :)