Ralim/IronOS #323: How to go to stanby mode from solder mode? 2 buttons do not work

jpis, 2018-06-29:
I'm submitting a Bug report. This could also be a feature request if what I describe is not supposed to work, which would surprise me.

The current behavior is:
In solder mode, I want to put back the TS100 to idle mode. I press both button, nothing happens

The expected behavior is:
In solder mode, press both buttons, the solder goes to idle mode immediately

***Steps to reproduce:***

1. Power the TS100
2. Press the A button 
3. Wait for the temperature to stabilize
4. Press both buttons

What are you running:  I downloaded the latest French version from GitHub Yesterday (June 28th)

  - Firmware Version: 2.04.1
  - PCB Version: (1/2) 2
  - Power Supply (Voltage and Current Rating) : 20V / 2.34A

jpis, 2018-06-29:
Moreover, 
1) I installed the EN version, this solved the problem. 
2) I re-installed the FR version problem was back.
3) Back to the EN version, problem solved. I keep this for now.
Seems that the FR release is flawed

jpis, 2018-06-29:
Problem was between buttons and chair. Sorry.

closed 2018-06-29 by jpis

Mrkvozrout, 2018-07-03:
You can use long press B (button away from tip) to exit soldering mode. It is easier than pushing 2 buttons.