Ralim/IronOS #327: Lithium cells cut off voltage level too high.

DrTeeth707, 2018-07-08:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ ] Bug report
  - [x] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Bug report and feature request. It's a bit of both.

* **What is the current behavior?**
Using 4S (or whatever the letter is) to "tell" my TS100 that I am using 4 18650 cells.

* **What is the expected behavior?**
To switch off the iron when the voltage is too low. It is way too severe. If you take the generally accepted, AFAIK, lowest permissible voltage of 3v, the cut off should be at 12v. It is much higher than that and the voltage has not dropped significantly since I started using them. I am going to use DC when using 18650s, especially as I have read some papers that the voltage can go much lower than 3v without significant harm.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. 
2. 
3. 

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**
To correct the low voltage cut-off levels.

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.x
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :

* **Other information** 

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

DrTeeth707, 2018-07-08:
Sorry, using the latest 2.04.1 version.

Ralim, 2018-07-08:
Hi,

There has been discussion on this previously, I won't repeat it all. But mostly it boils down to the use of 3.3V per cell is done because we don't know if the pack is protected or not, and it doesn't cut out a huge amount of energy since the cells start to drop fairly fast around this point under the load if you have small cells.

Currently it uses 3.3*number of cells. But note that this is a latched measurement, so if your cells are below this threshold for more than a period of time it will turn off. This often can be tripped by using cells with a high internal resistance that can sag under the load of the iron, or if you use thin or sub standard wiring as well.

DrTeeth707, 2018-07-08:
Re: [Ralim/ts100] Lithium cells cut off voltage level too high. (#327)

Hi Ben,

I would assume that the TS100 users are more savvy technically than other soldering iron users. Whilst the firmware plays safe, would it not be possible to have some extra options? That way the end user could run the risk of ruining his cells or not. I prefer to live dangerously.

Cheers

Hi,
There has been discussion on this previously, I won't repeat it all. But mostly it boils down to the use of 3.3V per cell is done because we don't know if the pack is protected or not, and it doesn't cut out a huge amount of energy since the cells start to drop fairly fast around this point under the load if you have small cells.
Currently it uses 3.3*number of cells. But note that this is a latched measurement, so if your cells are below this threshold for more than a period of time it will turn off. This often can be tripped by using cells with a high internal resistance that can sag under the load of the iron, or if you use thin or sub standard wiring as well.

JohnEdwa, 2018-07-08:
@DrTeeth707 
> if you take the generally accepted, AFAIK, lowest permissible voltage of 3v

Not "A safe and sensible cutoff voltage", the "lowest permissible voltage". It's the voltage at which you *need* to stop or you are risking permanently damaging a regular LiPo cell, often called a "Critical voltage". Especially when the TS100 cannot disconnect itself from the battery completely, just put itself to a sleep mode that still draws some current and will eventually completely drain the battery.

And like Ralim said, the difference in capacity between 3.3V and under is negligible. Even on good quality 18650 cells that can be taken down to 2.5V, it's mostly for added cell lifetime and safety, you still have almost all of the capacity between 4.2 and 3.3V. For the common Samsung INR18650-25R for example, the 3A discharge graph (max a TS100 will ever draw) [looks like this.](https://lygte-info.dk/pic/Batteries2012/compare/Samsung%20INR18650-25R%202500mAh%20(Green)-Capacity-3.0L.png) You are getting 2100mAh at 3.3V and 2400mAh at 2.8V, the difference is 14%. [At 0.5A](https://lygte-info.dk/pic/Batteries2012/compare/Samsung%20INR18650-25R%202500mAh%20(Green)-Capacity-0.5L.png) you are missing just 9%

I feel like 15% less capacity with 18650's is a good compromise so everyone using LiPo cells doesn't accidentally ruin them by draining them accidentally to 2.5V/cell. After all, you are getting more lifetime cycles out of your 18650s as well as a bonus.

closed 2018-07-27 by Ralim