Ralim/IronOS #330: Unable to flash successfully

ethan4, 2018-07-13:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
I've tried flashing multiple versions on both a Windows and OS X machine and in both cases I just get dots all over the screen and unable to turn the iron on, booting into DFU mode (3.43) works fine.

Both computers have USB3 only ports (Mac is USB-C) but connecting to a USB2 hub doesn't seem to help either.

* **What is the expected behavior?**

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. 
2. 
3. 

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

Windows 10 and Mac OSX 10.13.5.

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.x
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :

* **Other information** 

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Ralim, 2018-07-14:
Hi, can you confirm which language you were trying to flash on your iron?
sorry to hear about this, but more information will help.

When was the iron purchased?
Which exact versions have you tried?
if you haven't tried English can you please try it?
Something may either be broken or different on your iron for it to just be showing dots on the LCD (LCD did not finish initialisation properly).

Have you tested from under Windows (yeah I know it's a pain sometimes).

Ports being usb3 doesn't matter as the iron only syncs at usb 2.0 full speed.

ethan4, 2018-07-14:
Hi Ralim,

Trying to flash EN. 
I’ve tried the two most recent versions and 1.17. 

The flashing itself seems to be working fine as I get the mass storage device disconnecting and reconnecting with the file named to .RDY under both Mac and Windows. 

The dots on the screen seem to come back on the official firmware as well when I hold the second button (close to the screen) and click through some of the settings pages. The next also flips backwards and goes off the screen a bit. 

With the official firmware I can turn the iron on and control the temperature as expected. 

Maybe I have a bad unit?

Ralim, 2018-07-14:
It's possible it's bad, there hasn't been issues with the LCD yet, so it's probably time for something to come up.

My firmware runs all the hardware on a much tighter scheduler so it tends to expose any issues in the hardware.

ethan4, 2018-07-14:
Any suggested next steps here?

Ralim, 2018-07-14:
Honestly as of current my thoughts are a defective unit. Could have a dodgy solder joint on the screen or a damaged filter cap for the oled drivers. :/

Not really sure what would be the best options here.

closed 2018-07-14 by ethan4