Ralim/IronOS #332: Power issues with Ralim firmware

markc1984, 2018-07-16:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Not sure if it's a bug or a problem with my iron.

* **What is the current behavior?**
When connecting the power cord, absolutely nothing shows on display. Tried reflashing firmware, always succeeds with .RDY. Tried numerous versions of the firmware,up to the latest 2.04.1 with no success, screen is blank.

I'm experiencing exactly the same behaviour as reported on this page:
https://github.com/Ralim/ts100/issues/254

I tried the firmware posted towards the bottom of the page of the above link. No text on screen displays but pressing the A button seems to heat up the tip.

* **What is the expected behavior?**
Screen should display something, however nothing ever displays. Can't operate the iron without it with this firmware.

Just to confirm that this iron works with the manufacturer firmware (2.18 up to this writing), although sometimes I notice the display seems to go a bit weird when browsing the menu, like the display text will inverse back to front, or will just show a load of dots and I have to reconnect the device to the power supply.

* **What are you running:**

  - Firmware Version: 2.04.1
  - PCB Version: (1/2): Not sure
  - Power Supply 14v bench supply that can supply up to 7A

Ralim, 2018-07-16:
Hi,where is your unit from?
Honestly this is strange since the last issue before this one was about similar issues, but otherwise this is something I haven't heard of. If you are getting dots or screen inversions on the stock firmware then I would suspect either a dodgy screen or your iron has something else not quite right.

Their screen comms are bit banged in software and last I checked didn't actually check error codes, where ss I use hardware so if the screen throws an error it with not perish through it.

markc1984, 2018-07-18:
Hello,
I have received another TS100 from an alternative source and your firmware works perfectly. I can only conclude that as you have mentioned, the device is faulty. It is very strange how it would only work on the stock firmware though. 

Either way it will be sent back to the seller for a refund. It was from this eBay seller: https://www.ebay.co.uk/itm/12-24V-Mini-65W-Programmable-TS100-Electric-Solder-Soldering-Iron-LCD-Digital/262936921159?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

Thanks for all of your hard work in producing this firmware.

closed 2018-07-27 by Ralim

cebrax, 2020-01-24:
Hello Ralim,

I have the same issues with markc1984.. My iron was hit by ESD..

However, I don't have the budge to buy a new one..

I really liked your software and would love to use it instead of the dodgy original one.. Could you please create a version where you don't check the OLED with hardware so when the screen throws an error you just continue?

Ralim, 2020-01-24:
Hi,

This is because this firmware uses the hardware i2c peripheral rather than a software implementation (that had bugs).

Does your firmware work with the stock firmware?

What model of iron is it? (You haven't specified).

Does the stock firmware work fine ?
If so, which stock firmware ?

cebrax, 2020-01-24:
Hello Ralim,

Sorry for the miss of information.

It is a TS100 with DFU 3.43.

My iron works fine with Mini 2.18 firmware, however the screen has sometimes inversion of the text or some dots when in menus.

whitehoose, 2020-01-24:
Interested - cebrax when you say hit by esd, what were the circumstances?
The TS80 is an aluminium case which you'd exect to protect it, but while the 100 is plasticky (ABS?) which just happens to be an even better defence against standing pixies because it slows the discharge which is the aim of the game.  That's before you consider the suitability of ABS as a Tesla cage.  It's only it's tendency to melt and drip that stops it being adopted by the military. 
The blurb says It is designed with safety circuit and **anti-static structure** I don't like agreeing with wild claims - but this time I have to.  Latops, most caged PSUs and almost all domestic devices in an enclosure are pretty much sound round static.  The damage usually only gets done when people open the box and tickle the comonents.

I had the pleasure of "over volting" my TS80's power supply (trying out alternative QC3 power sources) and I got a bad dot situation as a result.  https://www.youtube.com/watch?v=yV8mPNVAUx8.
But I often expose my ts80 to quite high static and I've never had any problems, I believe both irons share the grounding screw arrangement so both have continuity.

cebrax, 2020-01-24:
Hello whitehoose,

I guess "ESD" is not the right term we should use here, sorry.. 

I was in a lousy factory environment and the earthing on the metal table I was working on was not great.. I also was getting tickles on the parts of my hands that touched the table when I was soldering... My TS100 was powered by a 19.2V laptop charger.

It may be possible that my iron got hit by a ground loop of about 60V?

whitehoose, 2020-01-24:
Spikes are more common than we expect, and they'll kill anything in their way.  I've just lost a trusted router when I accidentally "popped" the RDC in the garage.  Just bought an electric car - so now I have to worry about wiring installed to run heavy tools running spike free.