Ralim/IronOS #333: Boot Logo problem on v2.04.1

GootyS, 2018-07-16:
Hi Ralim or Anyone

Found a thread from last year where guys were having issues with a boot logo and some got it working. On v2.04.1 i tried what was suggested from the 2017 thread with no luck. What im getting is that soon as i add the logo.hex file the TS100 is dead, as in it wont power up. Soon as i reflash the latest v2.04.1 firmware, the iron works again.
Any ideas what i can try to get a boot logo onto the TS100? I used the logo editor on both pc and mac with the same result.

Ralim, 2018-07-16:
Hi,you are using an old version of the logo generator.
The address that was used in the first rev was incorrect, and so you need to recreate the hex file with the latest tool.

GootyS, 2018-07-17:
Thank you. Do you have a link to the correct one?

Ralim, 2018-07-17:
[Windows tool Here](https://github.com/Ralim/ts100/releases/download/v2.02/TS100.Logo.Editor.exe)
Or you can use the python script in the main repo.

GootyS, 2018-07-17:
It worked perfectly, Thank you so much

CornyjK, 2018-07-21:
@GootyS you may want close your issue.

closed 2018-07-27 by Ralim