Ralim/IronOS #334: Does this firmware work on the SainSmart Pro32 too?

reibuehl, 2018-07-17:
Is there anybody who has already tried this firmware successfully on the SaintSmart Pro32? The Pro32 seems to be a clone of the TS100.

Ralim, 2018-07-17:
It has worked in the past without issues,
Though it seems some units might be showing up more recently with OLED issues, but there should be no harm in testing it.

reibuehl, 2018-07-17:
Thanks! I'll give it a try.

closed 2018-07-17 by reibuehl

G-Pereira, 2020-06-24:
I'm having trouble flashing v2.10.1 to a PRO32, is there any known issue?

Ralim, 2020-06-24:
The PRO32 is a clone so it _should_ work but isn't something I guarantee as the schematics and parts they are using are not public and so I have no ability to ensure compatibility.

In terms of flashing the appear to clone the miniware bootloader so all the usual issues with the bootloader apply (is needing to double flash).

G-Pereira, 2020-06-25:
#11 fixed it for me. Cheers!

makergamer, 2023-05-28:
Just updated my stock SainSmart Pro32 to using IronOS v2.21. Did it the first try successfully:
* rename v2.21 TS100_LG.hex to PR32APP.HEX
* hold Button A (closest to tip) and plug in. It appears as 8 character drive on Windows 10
* copy PR32APP.HEX to it. wait a moment and the file manager will replace it with PR32APP.RDY
* eject USB storage and plug in again to confirm operation.