Ralim/IronOS #338: Bad display of accented letter in French 2.05 release

jpis, 2018-07-27:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [X] Bug report
  - [ ] Feature request
  - [ ] Translation

* **What is the current behavior?**
All French text having accented (stressed) letters are not well displayed. This is for the menus, the scrolling helps and the warning messages.

For instance: I see "Soudure Paramã"tres" instead of "Soudure Paramètres". Btw "Paramètres soudure" would be better.

* **Other information** 

I'll use the English version until this is fixed
Ralim told us he was not sure about Russian version in the last release because he was not using the Linux computer where he usually produce the releases. This is probably the same issue.

federck, 2018-07-28:
Also other releases seem to be affected by the same problem due to https://github.com/Ralim/ts100/commit/3d1d47aceeeb9ac20fc42e807fa19f326d7300f2 commit.

Ralim, 2018-07-29:
Hi, 
Can you test this build?

It appears that something went haywire with the encodings on the translation file.

if you can confirm this no-longer presents the issue i would appreciate it

[TS100.zip](https://github.com/Ralim/ts100/files/2238424/TS100.zip)

jpis, 2018-07-29:
Hi Ben,

 

This build displays good French. I keep it.

 

Best regards,

 

Jean-Paul

jpis, 2018-07-30:
Issue solved

closed 2018-07-30 by jpis