Ralim/IronOS #340: Power output as bar graph rather than percentage value

corbolais, 2018-07-28:
Graphical representation is easier and quicker to grasp than an abstract numerical percentage value.

* **I'm submitting a ...**
   - [X] Feature request

* **What is the current behavior?**
  - Power output displayed as percentage value: 65%

* **What is the expected behavior?**
  - Power output displayed as bar graph, i.e. as seen in [JBL power bar sample](https://www.batronix.com/images/generated/jbc-modular-line-di-d-783x797-White-min.jpg).

corbolais, 2018-07-28:
related issue #31

Ralim, 2018-07-28:
Hi,
that feature was in the 1.x firmwares and was not bought forward into the 2.x firmwares because its usability was fairly terrible, since there isn't a high resolution display on the unit and a large portion of the time it was either tiny, or completely full.

So when I did the re-write I didn't port this feature over as it didn't provide any more indication that just having heating on indicators.

corbolais, 2018-07-28:
Ok, thx for the clarification.

I'd favor a port, though. To me it is important to not only know the binary "heating on" but to know how hard (read: how much) the tip gets driven by PWM or simply power output for that matter.

It is an issue to be able to tell whether it is heating and at the same time have a quantitative guess at how much heat has to be "generated", as in "right now".

corbolais, 2019-10-09:
Hi,
I'm not sure why this got closed. I'm still interested in the feature, others might be, too.

I could get interested in implementing it, with a little heads up to speed up the learning curve of the src code organisation.

Regarding the low resolution, we could do with a horizontal 1px wide bar. That way, there's much more resolution available.

What do you think about it?

cheers

Ralim, 2019-10-09:
> Hi,
> I'm not sure why this got closed. I'm still interested in the feature, others might be, too.
Yeah shouldn't have closed this one :/
> 
> I could get interested in implementing it, with a little heads up to speed up the learning curve of the src code organisation.
> 
> Regarding the low resolution, we could do with a horizontal 1px wide bar. That way, there's much more resolution available.
> 
> What do you think about it?

I'm not particular on size, but it would be desired to work on both soldering screen layouts i suspect.

The code to modify is here: `workspace/TS100/Core/Src/OLED.cpp`

I would add a function to the OLED driver to draw the bar, that either takes in dimensions for location and %age, or hard code location if the same location works on both screen layouts.

Note that at the moment I haven't built out functions for setting individual pixels in the screen buffer as it hasn't been required yet. But this code can be taken from any other oled driver basically.

Once a funciton for drawing to the OLED has been made, hook it in like[ this function](https://github.com/Ralim/ts100/blob/c1db22cb4aa2e15610e5efc1ed0f339b938560bf/workspace/TS100/Core/Src/GUIThread.cpp#L512) does to calculate the remaining time.

discip, 2022-04-23:
@corbolais
Have you tested the latest release?
Do you still need this feature?
If not please consider closing this issue? 😊

thanks

corbolais, 2022-04-29:
@discip Hi there, haven't tested the latest release yet. Some news about this feature? Basically, I'd still like to see it implemented, I'm just not getting along time-wise.

discip, 2022-04-29:
No there are no changes in this regard, but maybe you are pleased with all the other new implementations enough to come over the need of a bar graph. 😊

discip, 2023-01-05:
@corbolais
Have you had time to test the latest release?
Especially with **`Detailed solder screen`** activated?
Maybe this will make you happy enough to close this issue for good. 😊

corbolais, 2023-01-08:
@discip Thanks for the ping.
Haven't tested it yet, though. But sounds great!
Closing for now.
Thanks again.

discip, 2023-01-08:
@corbolais
I really don't want to bother you ... but the issue is still open. 😀

River-Mochi, 2023-03-24:
@corbolais   

- you have to click the purple button at the bottom to  close this ticket.
- if you go into  User Interface >  and turn on Detailed Idle and Detailed Solder, you will see a lot better information you might like.
- if you get a Pinecil V2 iron, it has Bluetooth BLE,  then could use  Joric's  BLE API  on a browser  or  PineSAM app and they both have very nice displays of Power to watts and can be used to also control pinecil iron at the same time.
-  since the iron screens are very limited in size,  I think the alternative  is BLE bluetooth Browser apps which people can view in Huge size on Monitor  or  like I do on phone which is plenty big for me and I tap on the phone to change the Pinecil iron Temperatures. (only for newest V2  pinecil though has BLE).
- the BLE  apps are pretty fast at updating with V2 pinecil so it's rather fun to use in Color and  any kind of graph or bar could be implemented. Joric's  BLE Api  is a graph  and the PineSAM  app  can completely control Pinecil V2, but doesn't have a graph yet. 
- anyone is  able to help code in PineSAM to help implement  a power bar graph as well  just submit a PR.

River-Mochi, 2023-03-24:
This is an example of  exactly how PineSAM  BLE bluetooth app looks on my phone when I'm running newest Pinecil V2 soldering Iron.

[image]

closed 2023-03-24 by discip