Ralim/IronOS #342: Encoding error in DE release

pw-sys, 2018-08-01:
* **I'm submitting a Bug report

* **Do you want to request a *feature* or report a *bug*?**

* **What is the current behavior?**
In the DE release all Umlaute like äöü and ß are broken in the settings

* **What is the expected behavior?**
showing either äöü or ae oe ue

***Steps to reproduce:***

1.  Upload FW
2.  Rebbot
3.  goto settings
4. The second screen is already wrong

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**
It is anoing

* **What are you running:**
2.05 on PCB 1

Ralim, 2018-08-02:
This showed up in a couple of firmware builds, working on a new set to be put up shortly :)

Ralim, 2018-08-02:
Can you please test the 2.05.01 release ?

pw-sys, 2018-08-02:
Hi,

the 2.05.01 is not working at all. Tested DE and EN version. After the flash a .RDY file apears but after unpowering the iron the display stays dark. When I press the front button then it enters DFU-mode.
Going back to 2.05 still works

Trezona, 2018-08-02:
I also have this same problem with 2.05.01.

JohnEdwa, 2018-08-02:
Can confirm, 2.05.01 is fubar.

Lefinnois, 2018-08-03:
Same here with FR.
2.05 = utf8/iso8859-1 problem
2.05.01 = "DFU-mode" on screen when front button pushed

Ralim, 2018-08-05:
Pushed up new re-builds of 2.05.01.
Can you test and confirm ?

Lefinnois, 2018-08-06:
re-builds of 2.05.01 is ok now. thanks for your work.

closed 2018-08-08 by Ralim