Ralim/IronOS #343: Release v2.05.1 EN does not work here

bobafetthotmail, 2018-08-04:
I've just received a TS100 which appears to have PCB version "V2.5B" on the board and bootloader "DFU 3.43" ( I guess that's the bootloader version).

I tried to flash the latest v2.05.1 of your firmware TS100_EN.hex from a Windows 10 PC, and it seems to go well (the file changes name to .RDY as expected).

But when I pull the USB and plug in the power supply (SFF-desktop-grade 19.5 V 7.1A power brick), it does not boot.
If I press the A button I see DFU 3.43 on screen and that's all it does.

Flashing the stock firmware 2.18 or your v2.05 firmware TS100_EN.hex works fine and when I plug the power supply it boots the firmware and I can use the device.

Padre653, 2018-08-04:
Same issue here. 2.05 works fine, 2.05.1 will not boot.

bobafetthotmail, 2018-08-04:
apparently there is a bunch of reports of v2.05.1 not working in this issue https://github.com/Ralim/ts100/issues/342

Ralim, 2018-08-05:
Pushed up new re-builds of 2.05.01.
Can you test and confirm ?

closed 2018-08-06 by Ralim