Ralim/IronOS #345: auto shut-off function error

Roman26rus, 2018-08-16:
I'm giving ...

[X] Error report
What is the current behavior? I'm giving ...

[X] Error report
What is the current behavior?

active mode.
The soldering iron works well, but when the auto shut-off function runs out after the set time, the soldering iron switches off and does not turn on any more.

In the version of EN 2.04.1, 2.05, this function works. But in versions of RU, with Cyrillic this function does not work and leads to the soldering of the soldering iron forever.

the user needs the correct input.
What do you use:

V2.05 PCB2
DC input, Vcc = 19.2 V (maximum 2 A)

paulfertser, 2020-06-21:
Is this still reproducible? I just tried current code with lang=RU on my v2 board and OLED was turned off as expected. I suggest to close this ticket and let the reporter re-open if needed.

Ralim, 2020-06-21:
@paulfertser Indeed this should have been closed _a long time ago_.

closed 2020-06-21 by Ralim