Ralim/IronOS #347: Screen Glitching

rma6, 2018-08-22:
After some while without using my TS100 the screen started glitching.

When I power it, it starts just fine, but after a while the screen starts glithing. Here's a video: [https://photos.app.goo.gl/9ZZixrKs5km7PgQj8](url)

But if I plug it in the computer through the USB port the screen stays fine.

Does some one knows what I could do to fix the problem?

Ralim, 2018-08-23:
Hi,
In your video it appears that you are *not* running this firmware, but are running the Miniware firmware instead.

I haven't been tracking their recent firmware changes so not sure if they have changed anything.

If the glitches do not occur when running in the DFU mode then it is most likely something in their software that is causing something to fail. Mild hardware damage to the screen can cause this, where some screens seem to fail bad with stuck pixels / fail to update properly when the internal communications bus is run at higher speeds.

rma6, 2018-08-23:
I'll try to use this firmware and will report any changes.

rma6, 2018-08-29:
I tried to upload this software to my TS100 but I got this error message:
"File is too Large for Destination File System"

What should I do?

Ralim, 2018-08-30:
Hi,
Most likely you have not placed the unit into DFU mode.

rma6, 2018-08-30:
I am sure I have, I got a "DFU:3.43" message on the screen

rma6, 2018-08-30:
I'm using Windows 10.
No Anti Virus.
I've tested other USB ports.
I'm trying to load TS100_EN.hex (most recent version)

Ralim, 2018-08-30:
I'll have to get back to you more when I get home and can double check with mine.
But...

Usually its just a cast of boot into DFU mode, and then using windows explorer copy the .HEX file from your downloads folder to the drive for the soldering iron. Do **NOT** use any other tools than windows explorer for this. Do **NOT** download directly to the device. 

The file copy must occur in order because of how they coded the bootloader, nothing I can do about it really.

Ralim, 2018-09-17:
@rma6 
Closing issue due to no further response in > 14 days.

closed 2018-09-17 by Ralim

beshrkayali, 2018-11-13:
Hi @Ralim , I'm running into this same issue:

I'm on:
- Windows 10
- No anti-virus
- Tested multiple ports
- and latest release of this firmware.

Tried to copy/paste and to drag the file directly onto drive, always ending up with `File is too Large for Destination File System`.