Ralim/IronOS #348: Pressing both buttons is error-prone

skorokithakis, 2018-08-24:
* **I'm submitting a ...**
  - [X] Bug report
  - [ ] Feature request
  - [ ] Translation

* **What is the current behavior?**

Pressing both buttons is mistaken for a single-button press most of the time.

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.05
  - PCB Version: Not sure, 2 I guess
  - Power Supply (Voltage and Current Rating) : 24V, 10A.

It looks like the two-button press is implemented as a check for pressing one button a short time after the other button has been pressed. Instead, it would be much more user friendly if it were implemented like so:

On button down, check if the other button is also down. If so, consider it a double-press. If not, continue waiting for the button to go up.

This leads to much more user-friendly behaviour, as it detects double-presses every time, and the user can more easily press one button and then the other to register as a double-press.

Ralim, 2018-08-30:
Hi,
The logic I have implemented is the exact inverse of yours, but your point still works perfectly fine.
I'll have a fix for this merged into mainline along with the improved ADC and TS80 support Im trying to smash through now

skorokithakis, 2018-08-30:
Hmm, out of curiosity, what's the current logic?

skorokithakis, 2018-10-01:
Is there any way I could try the code on this? I'm really struggling every time :(

Ralim, 2018-10-07:
Yep I can make you up a build when I get back home from being away.

In most situations there are alternate exits rather than both buttons (such as holding the rear button).

skorokithakis, 2018-10-07:
THAT WORKS?! Daaamn, you just saved me a *lot* of frustration. That's easier than pressing both buttons, thank you!

Ralim, 2018-10-07:
Yeah, because the double press was hard for people I added it where that wasn't already used for something else :)

leeph, 2018-10-25:
Thanks for raising this @skorokithakis - this was my single biggest complaint with the Ralim firmware. I'm literally so excited right now I'm going to power up my TS100 just to switch it off with the rear button :)

Really love the firmware in all other respects - thanks @Ralim !!!

closed 2019-10-09 by Ralim