Ralim/IronOS #349: TS80 [TS200] Support

Ralim, 2018-08-28:
* **I'm submitting a ...**
  - [ ] Bug report
  - [X] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Feature, adding TS80 build support

The TS80 runs very similar hardware to the TS100 iron. The firmware does not need many changes to support this.

** Changes To do / Progress **

- [x] Pinout updates to support moved pins
- [x] Update ADC Code to use the newer pins
- [ ] Update ADC Scaling to support different power input ranges
- [ ] Update tip gains for the TS80 tips
- [ ] Implement Quick charge 2 & 3 support
- [ ] Test the new bootloader to see if it can be backported

cnnblike, 2018-08-30:
Hi, I'm thinking about porting the STM32 USB-PD (Power Delivery) to ts80, but I'm not quite sure about the hardware of TS80. I mean, will the hardware of TS80 support 20V 2A?

Ralim, 2018-08-30:
Hi @cnnblike ,
Sadly the TS80 does not support 20V (will die around 16V).
AND
The connections are not made for the USB-PD pins (No CC Pins).
There is only wiring to the D+ and D- pins.

I have already poked around to try and see about PD since I have more PD than QC chargers, but nothing can be done on this hardware sadly :(

cnnblike, 2018-08-30:
Oh, thank you for your reply @Ralim 
That's a pity. Thought this may work.

Ralim, 2018-08-30:
Yeah :( I can dream of everything being neat like this.
I'll keep on the manufacturers case to try and push for a USB-PD model, but I feel that it might be a ways off.

Though, a really nice product if your up for it, that is on my to-build list but buried too far deep would be to make a small USB-c inline adapater that converts between USB-PD and Quick Charge 3.0.

If you have interest in making that ^ hit me up via email and i'm happy to have a yarn about it.

cnnblike, 2018-08-30:
Maybe sometime when I had more free time :D
But think about it, maybe building a TS12 tips compatible, 3d-printable, USB-PD/QC3.0 compatible iron will get much more attention and be more attractive for the most hobbyists.

willumpie82, 2018-08-30:
something like this?
https://hackaday.io/project/18899-rt-soldering-pen

Ralim, 2018-09-03:
@cnnblike I agree, I would love a TS80 shell that can take even TS100 tips with USB PD. Maybe we will get it soon if they like us all lol

Ralim, 2018-09-04:
Progress is happening on this FYI to anyone reading, I just edit the top comment to track progress. Getting towards a releasable build soon.

ge0rg, 2018-09-08:
Would it be possible to use this with a regular 12V/9V supply instead of QC, e.g. by detecting a higher voltage on the input?

Ralim, 2018-09-08:
Yep, already supported.
Note that going much above 9v will pull more than 18W and I haven't tested how the tip and tip connector will last.

JohnEdwa, 2018-09-10:
The Weller WRMP pushes 40W through a similar 3.5mm connector, so the ~30% overclock from running the TS80 at 24W from a 12V supply *shouldn't* be a problem, though I do understand it's what's inside the plug that actually matters.
If it is though, implementing a power limit by calculating the maximum PWM duty cycle to keep it under 20W shouldn't be that complicated I'd reckon.

Ralim, 2018-09-10:
The changes over at #275  will come into play here nicely.

My biggest concern is just peak current rating and if the connector starts to heat, since if it heats it becomes a more significant drift in the temperature measurement.

I'm going to keep the QC code negotiating for 18W as thats the max for most adapaters (all?). However I wont stop you feeding in more. Note that some of the circuitry is only rated to ~16V though.

12V should be a nice point though. Also interested to look into the options for taking the voltage up higher anyway and just having a more stable PID.

JohnEdwa, 2018-09-11:
I thought I remembered it right, the stock firmware [has the power setting from 16W to 24W](https://youtu.be/_Z9es-D9_8g?t=701), even though QC3 only supports up to 18W. But it would indicate the iron can handle 12V no problem.

Ralim, 2018-09-11:
Nice catch, I never had the original firmware work well on my unit so I haven't been through it thoroughly yet. I have also had confirmation that 12V 2A is supported by the unit which is really good.

Though negotiating for 12V may come later on in my firmware since I want to do the tip calibration and then push out a beta version.

joric, 2018-09-21:
Will it accept firmware update from TS100? Will it turn into a brick afterwards? Are there any TS80 bootloader-compatible firmware updates in the wild?

Ralim, 2018-09-21:
@joric if you flash the ts100 firmware it will not die, but the pinouts are only similar on the OLED. Nothing else will work.

Not that I know of yet.

If you want I can upload a build of my TS80 fork here for you, its only really waiting on time to tune in the tip temperature calibration.

Ralim, 2018-09-21:
Yeah, its currently just an issue where the temp cal is fairly far whacked out atm (like 30 C).

Cant do much else atm though, no time for development.

joric, 2018-09-21:
Apparently it doesn't even power on with PD laptop power supply, QC only. That's pretty bad, I thought I wouldn't need another power supply on the table. Also that 3.5 mm jack isn't too sturdy and tip wobbles, like a lot. Aluminum sleeve doesn't help very much, maybe it's worth to buy carbon one instead when it comes out. My TS80 unboxing video: https://youtu.be/xc-yIq0m54o

Ralim, 2018-09-21:
It is a QC3 device, it won't work with PD! QC is a proprietary protocol that is not compatible with the USB-PD standard.

The 3.5mm Jack shouldn't take the strain at all and the mental housing takes the force. The only TS80's I have seen have all been fairly snug with justa little bit of wiggle which is normal.

In my firmware I'm allowing 5V operation as a fallback which works with some PD chargers that don't follow the spec completely. 

Also their charger does the higher 24W support which is good.

andersvos, 2018-09-22:
This would be perfect for using the TS80 in the field https://www.indiegogo.com/projects/omni-ultimate-a-new-generation-of-power/x/4042493#/updates/all

joric, 2018-09-22:
@andersvos $16 Xiaomi 10000mAh power bank works just fine, I have one of those https://www.aliexpress.com/item//32826648959.html TS80 only supports 18W (12V * 1.5A) max.

andersvos, 2018-09-23:
@joric

The Omni has QC 3.0 too, and has more than double the capacity.

gbraad, 2018-09-23:
@Ralim would you have a test version of the firmware available?

joric, 2018-09-23:
@Ralim looks like we got stock 1.06 firmware hex from minidso (no sources yet but it could be helpful to restore TS80 to factory state). Direct link to file: [TS80.V1.06.hex](http://www.minidso.com/forum.php?mod=attachment&aid=NzA3NXwzMWQ5ZTA0NXwxNTM3NzQ1ODMxfDM2NDV8MzIwMw%3D%3D) (from [this](http://www.minidso.com/forum.php?mod=redirect&goto=findpost&ptid=3203&pid=6834) forum post).

Ralim, 2018-09-24:
@joric QC2 *should* work on their firmware from my reading.
QC 2.0 is implemented on my firmware, but I have nothing I can test with over here :(

Here is a rough build, will feel very similar.
Still needs temp cal.
Would love feedback on if QC works for you.
[TS80.zip](https://github.com/Ralim/ts100/files/2409546/TS80.zip)

gbraad, 2018-09-24:
@Ralim I tried the firmware, but the first time it started the OLED stayed black, but it would draw power when I pressed the button. After unplugging and replugging it showed v1.06 and it felt like it started the old firmware?

Ralim, 2018-09-24:
How did you try and do the upgrade ? Did you make sure to do it twice if it threw an error etc ?

gbraad, 2018-09-24:
I retried it 3 times, and eventually it worked and showed the `.RDY` file.. it didn't feel like it rebooted, but dismounted and shows the disk drive (not able to open)... I disconnected and reconnected and it works. It even heats on 5V from the laptop.

the flipping of the display seemed to fault at first, but it works... will come back with more info after some testing.

joric, 2018-09-24:
@Ralim worked for me right away. Temperatures seem about accurate. https://youtu.be/ngfiotSwQuA

Ralim, 2018-09-24:
@joric whats the stock cable like ? (In my sample I didnt get the nice cable)

Glad to hear it flashed ok :)

Yeah the accelerometer is mounted in a different orientation in the TS80, need to figure out which way round the rollover *should* be :)

I'm not sure about the screen, its centered on my unit but mines pre-production :| Would love to hear others feedback on this.

I'm planning to merge the TS80 fork into mainline so that translations can be updated, haven't finished work on the firmware but want to get something somewhat out there :)

- Ill keep this issue here for tracking progress on the TS80 support

joric, 2018-09-24:
@Ralim it's 1m Type-A/Type-C heat-resistant black silicone cable, very soft but I couldn't copy large files from my phone to USB 3.0 (I have no USB 2.0 ports) with it, everything freezes and stops midway.

joric, 2018-09-25:
@Zyzto I use PD-chargeable power bank for that (CDQ07ZM), roughly about the same :) There is X-CUBE-USB-PD stack, part of the STM32Cube software-development ecosystem, that can be downloaded, free of charge, from https://www.st.com/en/embedded-software/x-cube-usb-pd.html (also on github https://github.com/st-one/X-CUBE-USB-PD) not sure if it's possible to add PD support with firmware-only mod though.

Ralim, 2018-09-25:
@Zyzto  Havent tried but should work, and if it doesnt happy to try and figure out why :) 

@joric Please see the first comments in this thread. PD support cannot be added as the frontend required is missing (and there is no connections at all to the CC pins).

joric, 2018-09-25:
Miniware has published official TS80 firmware source code (currently 1.06):
* http://www.miniware.com.cn/product/ts80-soldering-iron-main-2
* http://www.minidso.com/forum.php?mod=viewthread&tid=3208

Ralim, 2018-09-25:
@joric Yeah hardware mods are what I'm trying to avoid. 

Honestly not worried about trying to get PD in the unit, but rather would inline an adapter :)

@gbraad Let us know what you find, lucky to live over there :)

Houseman74, 2018-09-26:
Hi to all.
Cannot understand why my TS80 can't be powered from a Blitzwolf 15000mA power bank that claims to have Qualcomm QC3.0 certified.
Every time I plug the iron in the dedicated QC3.0 USB port of the Powerbank the Iron goes off and displays Low Vot.
It seems the power bank cannot deliver more than 5V output, but plugged into a powermeter and a phone it recharges it in QC2.0 mode..
Any hints?
Best regards

Ralim, 2018-09-26:
Hi,
I cant really provide support for the official firmware, but is your power bank QC3 or QC2.0 ? Both should work but some units are more fussy than others. Not all "QC" banks actually follow the standard :(

Do you have an inline usb power meter that can show you the voltages on the D+ and D- pins?

Houseman74, 2018-09-26:
Hi Ralim 
Thank you a lot for your quick reply.
My power bank is following: BlitzWolf BW-P5
bought at:
https://it.aliexpress.com/item/BlitzWolf-BW-P5-15600mAh-Quick-Charge-QC3-0-Dual-USB-Power-Bank-Universal-Fast-Charging-For/32783530009.html
Everywhere it shows a full QC3.0 support.

Power meter shows D+ 1,66V D- 1,65V when connected to the powerbank QC3.0 port.

I saw one hour ago that mindso posted at this link:
http://www.minidso.com/forum.php?mod=viewthread&tid=3225&extra=page%3D1

an unofficial fork, but apparently supported by them in which it claims:
"On the basis of 1.06 firmware issued by the government , a firmware was made to 
solve the problem that TS80 solder pen can not be used under the mobile power 
supply. It was named 1.06s. The solution is as follows."

The firmware.hex file is 80kB and the fact is that I cannot load it into the ST32 Mass storage device since there is only 8kB in disk.
Am I doing something wrong??
Is the disk E: one that displays once inserted the TS-80 in the USB port of a PC right?
Best regards.
Steve

Ralim, 2018-09-26:
Looks like that power supply supports QC, but those voltages are fairly out (typically 3-3.3V is used).

That fork is to stop power banks going to sleep, it probably wont help your bank not working.

If you check the main page of this repo or the booklet that came with your iron, you need to hold the front button while connecting to the computer to boot up into the bootloader mode to let you update the firmware (It will say DFU on the screen). Then you can drag and drop the file onto it to flash it.

Houseman74, 2018-09-26:
Yes You are right. Sorry of course I have to start in DFU mode (as all the MINIDSO units)..
And Yes you are right, the upgrade does not fix the problem.
Low-Vot displayed also.
What does it mean. The power bank is not fully QC3.0 certified or is the TS-80 tight?
Regards

Ralim, 2018-09-26:
All good :) 

Not quite sure whats going on, can you try my firmware and check against it as well ?

Houseman74, 2018-09-26:
of course I will.
I let you know.
Thanks

Houseman74, 2018-09-26:
Genius!!!!
It works!!!
Man, you made my day!!!
Thanks a lot!!

Ralim, 2018-09-26:
No problem, the firmware has a few rough edges (temp cal isnt perfect, doesnt support >18W mode) but should be usable until i get a full release out soon :)

Houseman74, 2018-09-26:
Thaks pal. Just a few question:
what has been changed in the source to enable use of QC3.0 different from the original one?
Where is your main page for the new releases of TS-80?
I can only see TS-100 one..
why isn't the ST32M drive shown anymore once plugged in?
thanks

Ralim, 2018-09-26:
Entirely different code :) Basically the TS100 and TS80 are very similar in hardware and as such I extended the TS100 code to work on the TS80 (Thats what this issue here is to track my progress on). 

So the code that does the QC3.0 negotiation is totally different to theirs (though inspired by theirs, and using some of the docs on QC they sent me). 

The TS80 builds will show up in the releases tab alongside the TS100 firmware when I make a full release when all the features are working

There is no drive anymore, just the menu on the device :)

Basically, this code is build with the free GCC toolchain, which is not as good at optimising as IAR, so way back in the beginning of all this, I took out the virtual drive to make room for extra features instead :) But dont panic, DFU mode still works fine so you can still upgrade etc the same.

gbraad, 2018-09-26:
Off-topic; but so far I have only found a QC to PD fast charge converter. I hadn't spend a lot of time on this before, but it seems QC is a very common for the Chinese versions, so very understandable they also choose this for the TS80. I do not agree, but I understand the choice.

joric, 2018-09-26:
@gbraad it's getting a bit out of topic but I think one jumper wire (IF CC is not already connected on the rear side of the PCB) and firmware support is much better than a bunch of ugly adapters. Google for example hates QC and deprecates QC support in favor of PD (see https://www.androidguys.com/news/quick-charge-4-0-is-coming-but-google-might-kill-it).

gbraad, 2018-09-26:
to not to take it further off-topic we perhaps would have to discuss this somewhere else. (and yes, I also dislike having to take additional adapters, but soldering inside is also not really preferred).

Ralim, 2018-09-27:
Chinese power banks use QC a lot as phones use it, where as PD was really designed for laptops etc.

QC costs about 3 resistors (basically free) to support, where as PD is a much more complicated protocol (1 wire ethernet anyone). PD would require a much larger AFE or a dedicated PD chip.
PD isnt really going to be easy to modify into the existing units.

For example, here is ST's descrete AFE : 
[image]

So if we were to go to doing something like this, it would be better to try and mod a STUSB4500 IC into the unit.

QC is a ugly pain that *legally* requires you to pay licensing. PD is at least an open standard.

joric, 2018-09-27:
@Ralim There's software implementation by Reclaimer Labs (not sure about circuits needed for this, I thought it's enough to hook up MCU pins to USB pins) but it's [5000 lines of code, including a 1200 line state machine with 29 states that uses 70% of the Arduino program space](https://www.reclaimerlabs.com/blog/2017/2/1/usb-c-for-engineers-part-3), hell knows if it's doable but that PD looks like even bigger abomination than QC now :)

Ralim, 2018-09-27:
@joric 
I have the FSUSB interface that code uses. That code impliments the state machine but none of the signalling on the wire, so going with a descrete AFE (like above) requires even more code. Which is why I like the STM IC's instead for PD as they have the state machine inside them (but cost slightly more).

usb PD is a hell of a protocol, and takes a large amount of space if you try and program the state machine etc internally. QC is annoying because of the supposed secrecy around it, but PD is just massively complex because of all the things it supports. (PD is actually a subset of the functions that interface supports).

1461748123, 2018-09-28:
Is there any test firmware we can try on? I've been waiting for TS80's support the day it came out :)

gbraad, 2018-09-28:
Scroll up and you will find a zip file (a test file was released a few days
ago)

1461748123, 2018-09-29:
> Scroll up and you will find a zip file (a test file was released a few days ago)
> […](#)
> -- Gerard Braad | http://gbraad.nl [ Doing Open Source Matters ]

Thanks for the reply. I tried flashing it multiple times and keep getting .ERR file.
I'm not using the stock cable, I'm using a XiaoMi USB-C cable which is confirm working on my other devices.
Here's the .ERR file if it helps!

[TS80.zip](https://github.com/Ralim/ts100/files/2430128/TS80.zip)

Ralim, 2018-09-29:
Hi,
That is very common on some computers, this is not the firmware but instead the way Miniware implemented updates. Try coping the file multiple times in a row or using the python loader script etc.
This is documented elsewhere on this repo :)

1461748123, 2018-09-29:
> Hi,
> That is very common on some computers, this is not the firmware but instead the way Miniware implemented updates. Try coping the file multiple times in a row or using the python loader script etc.
> This is documented elsewhere on this repo :)

Thanks a lot! I tried again today and its working now. I will report back any issues related to the firmware!

gbraad, 2018-10-03:
Off-topic: currently resorting to the following solution:
[image]
This allows me to use USB-PD as the slim-tip cable requests 20V (this can be used for the TS100), and the DC-DC converters understands QC and a bunch of other fast charging protocols. This way I can also use any regular ThinkPad slim-tip to power both irons.

gbraad, 2018-10-03:
@Ralim is it possible to draw 12V as supply voltage?

Zyzto, 2018-10-03:
@gbraad in the Iron u can set off volt to 13 and I saw video where it reached near 11v so I guess if qc can hit 12v u can

[450 more replies]

closed 2023-07-21 by Ralim