Ralim/IronOS #350: OLED burns in => Turn OLED off

thezenox, 2018-09-01:
* **I'm submitting a ...**
Feature request

* **What is the current behavior?**
OLED stays in idle screen

* **What is the motivation / use case for changing the behavior?**
My oled "burned in"/different brightness on pixels when powered all the time

I have my TS100 pluged in all the time and the OLED starts to get dimer on the pixels turned on in standby mode. So it would be a nice feature to turn the display off (after adjustable time) and turn it on again with motion detection. Could also be done with custom idle screen (#190 ) if a blank screen is used

Ralim, 2018-09-01:
The TS100 code will already turn off the screen when the unit is sitting idle to prevent this *except* on the russian language (#345)

thezenox, 2018-09-03:
I upgraded to v2.05.01 and discovered, that it's not entering sleepmodes at all (neither standby nor shutdown). Last version working for me is v2.04.1 (4718efe). reset settings doesn't change anything. All using DE firmware using 10sec, 1minute and sensitivity 3 . 
Am I doing anything wrong here ?

Ralim, 2018-09-04:
Most likely its not you but the compiler, I've been finding small edge case issues that only show up on some langauges. Next build is going to have workarounds for this I *hope*

One thing that would be useful, is if you could enter the info menu (Hold rear button on the idle screen and info shows up). If you press through with the front button you can scroll through the various values. If you could write down these it would help narrow it down a bit :)

thezenox, 2018-09-07:
Here you go:
V2.05 PCB1
Heap: 01416
HWMG: 00650
HWMP: 00482
HWMM: 00424
(Is Move supposed to display the same as Time, but only updating when moving?)

Facing the same problem with v2.05.01_EN

Ralim, 2018-09-08:
You are correct in the move timer. It's the last time movement was detected. Your memory values are good so that's nice.

Good to know your on the older accelerometer. (They work the same, but slightly different sensitivities)

Is your move counter always incrementing?

torqu3e, 2019-05-30:
Just got my TS100, exact same firmware info as above and the move counter does not stop incrementing. It goes up in lock step with the time counter.

Is the accelerometer borked? though it does detect the orientation of the iron if I set it to ambidextrous.

Changing motion sensitivity to 0 through 9 doesn't make any difference.

Ralim, 2019-08-27:
Closing in favour of #366

closed 2019-08-27 by Ralim