Ralim/IronOS #364: Settings don't persist

dhiltonp, 2018-09-24:
Hash 056353ed works, 54ec20cd and after fail, this corresponds to PR #357.

detailedIDLE/detailedSoldering settings do not persist. I tried revbumping the settings and doing a reset via the menu with no change in behavior.

If I set the checkboxes, the idle and soldering screen do not go to advanced. If I go through the menu again, the checkboxes have reset to their default values. 

I thought there might be an indexing error, but I counted those two fields and they look fine.

I have not tested the persistence of other settings.

Ralim, 2018-09-24:
What is the flash size of the build on the non-working build ?
Most likely it is oversize and running into the flash page being used to store the flash settings.

jvitkauskas, 2018-09-24:
The only thing which could be at fault is that PIDSETTINGS support and changed array sizes because of that. Other than that there we no code changes (except translation text)

dhiltonp, 2018-09-24:
:/

I hoped it wasn't something like that. I thought we had 64k total, the build is 55288.

Ralim, 2018-09-24:
Uh no.

* 8K for bootloader.
* 1K for settings page.
* 1K for startup screen page.

Leaves 54K for program.

As your in the 55K area you will be overwriting the settings page and code on the same area of flash.

The linker script in this report should throw errors if you try and go over 54K?
(Let me know if it doesn't as that *is* a bug).

Ralim, 2018-09-24:
@dhiltonp Bugger, I'll have a look into repo'ing this locally.

If its easy, whats your local build setup ?

dhiltonp, 2018-09-24:
GCC 8.2.1 on Arch linux. I've just been running `make`

(well, except for a few calls to `build.sh` for verification)

Ralim, 2018-09-24:
Interesting, I dont have a GCC 8.x installed anywhere atm, but doubt it would break too much.

dhiltonp, 2018-09-24:
Here's the raw output of make:

[build.txt](https://github.com/Ralim/ts100/files/2409881/build.txt)

Ralim, 2018-09-24:
I have to revise my above comments, it is actually 16K for the boot loader on the TS series of soldering irons. Which is why the linker script has a ROM length of 46K.

Your build however looks like its and OK size (~44K for FLASH content).
Which is why your build is going OK.

Which means your compiler could be trying to "optimize" the flash write system if settings are not saving.

closed 2018-10-18 by Ralim