Ralim/IronOS #366: TS100 wont go into timed sleep or shutdown

3dBu, 2018-09-29:
* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **What is the current behavior?**
Soldering iron stays on, if I leave it powered and untouched on the table.

* **What is the expected behavior?**
Soldering iron should go to sleep mode and shutdown after set time.

***Steps to reproduce:***

1. Power iron
2. Turn it on
3. Don't move it
4. Wait for timed sleep & shutdown

* **What are you running:**

  - Firmware Version: v2.05.01
  - PCB Version: 1
  - Power Supply: 20V 4,5A (Lenovo laptop psu)

jonnieZG, 2018-09-29:
It has not been working since v2.05. I had to revert back to v2.04.1.

Ralim, 2018-09-30:
Can you please detail which firmware version you are using, and sensitivity, and what the M count in the debug menu does when both moving and not moving your iron, and what PCB model you have in the debug menu ?

I still have no irons that reproduce this issue.

jonnieZG, 2018-09-30:
It says V2.05 PCB1. Move counter keeps on counting seconds, the same as Time, no matter if I move it or not, no matter what the sensitivity value is.

Ralim, 2018-09-30:
Which language / exactly which file ? Cant reproduce this on either of my PCB1 boards on 2.05 / 2.05.01

jonnieZG, 2018-09-30:
Tried with EN too, still does not work.

- Heap: 01416
- HWMG: 00650
- HWMP: 00482
- HWMM: 00424
- Time and Move - both ticking the same seconds value

3dBu, 2018-09-30:
EN language for me too.

jonnieZG, 2018-10-01:
I have not tested it (and I won't get to any time soon), but the problem was introduced somewhere **after** the commit  https://github.com/Ralim/ts100/commit/4718efe79bfcebb91cf09a6331974938989d3230.

If you look at the main.cpp at commit https://github.com/Ralim/ts100/blob/4718efe79bfcebb91cf09a6331974938989d3230/workspace/TS100/src/main.cpp, you will see that the initialization of the datax array at that time was done properly using memset:
```C++
	int16_t datax[MOVFilter];
	int16_t datay[MOVFilter];
	int16_t dataz[MOVFilter];
	uint8_t currentPointer = 0;
	memset(datax, 0, MOVFilter * sizeof(int16_t));
	memset(datay, 0, MOVFilter * sizeof(int16_t));
	memset(dataz, 0, MOVFilter * sizeof(int16_t));
```
Later on, you started using the shorter version, which does not look kosher to me, since I suspect it to be leaving garbage in the memory, causing overflows in the calculations of the average:
```C++
	lastMovementTime = 0;
	int16_t datax[MOVFilter] = { 0 };
	int16_t datay[MOVFilter] = { 0 };
	int16_t dataz[MOVFilter] = { 0 };
```
Again, this is just my gut feeling and I did not go through the code thorougly, but perhaps you might consider using the EWMA filter for smoothing accelerometer readings rather than buffering last 8 readings: https://github.com/jonnieZG/EWMA.

jonnieZG, 2018-10-01:
Oh, and also this: you absolutely **should** be using the `volatile` keyword in definitions of all global variables that are defined by another OS thread. It prevents the optimizer from doing stupid things with variables that are being shared between several threads!

jonnieZG, 2018-10-02:
@Ralim, I have just added a template for this filter, called EwmaT, that does not use any floating-pointh maths. You can use it like this (the alpha and alphaScale values have to be empirically determined):

```C++
#include "EwmaT.h"
...
EwmaT<uint32_t> ewmaX(alpha, alphaScale);
EwmaT<uint32_t> ewmaY(alpha, alphaScale);
EwmaT<uint32_t> ewmaZ(alpha, alphaScale);
...
filteredX = ewmaX.filter(tx);
filteredY = ewmaY.filter(ty);
filteredZ = ewmaZ.filter(tz);
...
```

nmaggioni, 2018-11-08:
I'm having the same issue, and the debug values are [the same as above](https://github.com/Ralim/ts100/issues/366#issuecomment-425742068).

I've bisected the commits between v2.04.1 and v2.05, and the problem begun to appear in commit 215fe8e indeed.

Ralim, 2018-11-10:
@nmaggioni Could you confirm if the master branches fix in f4e2 helps for this ?

nmaggioni, 2018-11-11:
@Ralim It seems to be solved indeed! Thanks for the quick fix.

Ralim, 2018-11-11:
Thanks for checking, stupid typo that your bisection made easy to find.

jonnieZG, 2019-01-04:
Solved as of v2.06. Please close.

haarp, 2019-07-25:
Is it really solved? I flashed 2.06rc5 on my iron which I just received the other day, and it will not go into sleep or shutdown at all. Even setting sensitivity to 0 (supposedly off) doesn't help. The motion counter keeps ticking no matter how still the iron is.

Ralim, 2019-07-25:
@haarp 
Turning sensitivity off stops the system working with the IMU, and instead forces the counter to increment so the iron doesn't sleep (as otherwise you have no means to stop it sleeping during soldering).

What PCB model is your iron showing up as ?
What sensitivity levels have you tried ?

haarp, 2019-07-25:
Ah, I see, I'll avoid 0 then. Just for the record, automatic display orientation does work, so the accelerometer msut be functional. Running on an old 16V Thinkpad power supply.

> What PCB model is your iron showing up as ?

According to the debug menu: Model: 1 (btw. once I reach Tres in the debug menu, it stops responding to input)

> What sensitivity levels have you tried ?

From default 7 all the way down to 1

jakester21, 2019-07-30:
im having the same problem just updated to the latest firmware and the seconds never count down and im at default sensitivity

Ralim, 2019-07-30:
@jakester21 
If you are talking about the debug menu:
Not counting down is ok, as its a timestamp. So when movement is detected the value should increment, and when its sitting still it should keep the same number.

jakester21, 2019-07-30:
ok but it does not go to sleep the temp always stay at the temp i set it to and it does not shut down after the time i set but the v2.4.1 seems to work

fseuhs, 2019-08-03:
I have the same problem with a brand new TS100.
No shutdown and sleep with v2.06-RC5 and v2.05.01, I tested!
v2.04.1 works well.
Regards

meandrey, 2019-08-13:
I had the same problem with v2.06-RC5, the device wouldn't go to sleep/shutdown (tried different motion sensitivities).
I've also tried  v2.06-RC4 and it work's fine with this version.
I just got the device (1 hour ago) so still trying to figure it out hot to access Debug menu in order to provide more info.

yakimka, 2019-08-20:
I have the same problem with a brand new TS100.
"Move" in debug menu ticking like seconds.
Once I reach "Tres" in the debug menu, it stops responding.

Firmware Version: v2.06
Model: 1
Power Supply: 24V 3A

yakimka, 2019-08-27:
Yes, automatic screen rotation is working for me in 2.06-RC5

Ralim, 2019-08-27:
This is looking like a bug in the older acceleromter model used in the PCB 1 rev units.

I do have one of these daughter boards "somewhere" that I need to track down to solve this.

Sorry to be so absent these days, just really dont have much time these days.

Could someone confirm if automatic screen rotation is working for them?

Suspect file is here : https://github.com/Ralim/ts100/blob/master/workspace/TS100/Core/Src/MMA8652FC.cpp

th3Al3xander, 2019-08-30:
I can confirm the bug
Version 2.06

stuartm, 2019-09-05:
Hardly seems necessary to add yet another report, but on the basis that more data points can help confirm/debug, I'm seeing this issue on a 'brand new' iron with 2.06-RC5 flashed, Model 1 using a 19v supply. Auto-rotation works (though I turned it off) but movement value climbs at a regular rate (1 per second approx).

If I get time I'll do a binary search and try to identify the commit.

Firebie, 2019-09-05:
How to compile latest sources on Windows? I was using Atollic, but now, when I try to compile, I have
"Nothing to build for project TS100", when I open project properties - "Orphaned configuration. No base extension cfg exists for com.st.stm32cube.ide.mcu.gnu.managedbuild.config.exe.release.980996604".

evgazloy, 2019-09-05:
Confirm. 2.06-RC5. automatic screen rotation doesn't work either.

Neposedaua, 2019-09-26:
v2.06-RC5, the device wouldn't go to sleep/shutdown (tried different motion sensitivities).
Confirm.

Saschko2000, 2019-09-28:
v2.05.01.  I have two devices. One works like a charme, the other will not go in sleep mode. Both have the same Firmware. I have no idea :-(

Ralim, 2019-10-08:
[TS100_temp_3fea.zip](https://github.com/Ralim/ts100/files/3702041/TS100_temp_3fea.zip)

Heya,
This is a build from my branch that I'm working on for some better temperature calibration/temp curve work.

Could you please use the debug menu to see if the move timer is working for you in this version?
Also, as this is from my temp branch, would love if i could crowd source some data from you all :grin: 
-> Please do a temperature calibration in the menu.
This is done by letting the tip and handle come into temp equilibrium (ie, leave on desk and ignore for an hour). Then power up the unit, and go into the Settings menu > Advanced > Temperature calibration > press the front button (for OK). Then it will dance " . . . . " (about 1.5 seconds).
Then show a checkbox and a 4 digit number.

Would be wonderful if you could post back your tip type and the 4 digit number. Hopefully most are around 700-900 range if my maths works out :D

NOTE:
-> This version may not have perfect temp calibration atm (though i hope its moderately close).
-> Should have some minor fixes for quality of life stuff like the power estimator too.
-> *Might* be better at temp control depending on your tip.
-> Built from git tag `3fea95c`

If you find the temp control is way off, let me know.

th3Al3xander, 2019-10-12:
Engish is my 2nd language, so im sorry in advance if i make mistakes

> Could you please use the debug menu to see if the move timer is working for you in this version?

If i move the ts 100 it counts upward, if i dont move it it stops counting

> -> Please do a temperature calibration in the menu.
> This is done by letting the tip and handle come into temp equilibrium (ie, leave on desk and ignore for an hour). Then power up the unit, and go [...]

it says 673

Thanks for supporting and making the software

ggc161169, 2019-10-12:
I had the same problem, but I solved installing the version 2.04, then the 2.05 (nmaggioni) and then the 2.06 RC4 ... without problems so far.

Ralim, 2019-10-13:
@th3Al3xander 
Thank you, you did exactly what I wanted.
Looks like accelerometer is working for you (at least).

Also, that calibration number looks good, which is promising :) An offset of ~600uV sounds about right for the ADC + resistor calibrations.

Need to re-calculate some curves but this is promising

Firebie, 2019-10-16:
Small patch for undervoltage "V" symbol:
diff --git a/workspace/TS100/Core/Src/GUIThread.cpp b/workspace/TS100/Core/Src/GUIThread.cpp
index 2bdaaa5..fe10d76 100644
--- a/workspace/TS100/Core/Src/GUIThread.cpp
+++ b/workspace/TS100/Core/Src/GUIThread.cpp
@@ -193,7 +193,7 @@ static bool checkVoltageForExit() {
 				OLED::setCursor(0, 8);
 				OLED::print(InputVoltageString);
 				printVoltage();
-				OLED::print("V");
+				OLED::print(SymbolVolts);
 
 			} else {
 				OLED::setFont(0);

Yourigh, 2019-11-18:
> [TS100_temp_3fea.zip](https://github.com/Ralim/ts100/files/3702041/TS100_temp_3fea.zip)
> 
> Could you please use the debug menu to see if the move timer is working for you in this version?

Timer is working well as well as sleep.

> -> Please do a temperature calibration in the menu.
> This is done by letting the tip and handle come into temp equilibrium (ie, leave on desk and ignore for an hour). Then power up the unit, and go [...]

It shows 0753 on tip TS-B2
It shows 0827 on tip TS-ILS

Also turning of by pressing both buttons is much easier as in 2.04. Thank you!

Ralim, 2019-11-18:
Awesome!

Thank you everyone for testing this.

Any general feedback on stability of the firmware ^ 

I need to make up a new test jig to calibrate the temp curve but then aiming to merge that into master and get a release out.

closed 2019-12-29 by Ralim