Ralim/IronOS #369: Dead on Ralim but alive on stock fw...

rnelias, 2018-10-04:
Hi there, I have a two years old TS100 that suddenly stopped working on Ralim fw. At first, I've thought the iron was dead but after installing stock fw it can back to live but none Ralim version that I've tested until now is working anymore :( 

Any clue what could be happening?

The iron neither brings the boot screen...

Ralim, 2018-10-04:
Hi,
Next time please use the template, or else your issue will be closed.
I know parts of it are not relevant, but knowing what firmware you were most likely last on would help.

Does the unit go into DFU correctly?
Under the stock firmware does the screen and Accelerometer work correctly?
Have you done a full feature test under their stock firmware?
You say "Suddenly stopped working" Usually software doesnt suddenly stop, but rather hardware issues crop up that cause issues. 
What voltage are you testing on? Have you tested using USB power as well as the 2.1mm jack ?

"The iron neither brings the boot screen..." Doesnt really make sense?

rnelias, 2018-10-04:
I've used Ralim for about a year when suddenly the iron stopped booting up. I thought it was dead and even bought another one but after flashing the stock firmware it came back to life again. After this event, I've tried to use other Ralim versions but I only get black screen and nothing happens, however, if I return the stock fw the iron works fine. 

Nevermind... I'll keep using stock fw and you can close this issue. Sorry for bothering you. Have a nice day

Ralim, 2018-10-04:
I'm happy to figure out why, but you haven't given me anything to work with at all here.

I'm going to assume by doesn't boot up you don't get to see the screen at all ?

rnelias, 2018-10-04:
DFU 3.43 it goes back and forth into DFU mode without any problem
Stock fw 2.18 works fine
power supply is 20V with the correct plug that I have used for about 2 years (and keep using)

With Ralim fw, nothing happes. No boot screen... nothing...

I'd guess the iron does not have enough memory to run ralim fw...

rnelias, 2018-10-04:
I've been using this iron with stock firmware without any issues... heats up and accelerometer puts it in stand-by normally. 

I'll record a video showing it with both firmwares and send you the link

Ralim, 2018-10-05:
Its really odd this is happening for you.

My firmware includes a much larger number of checks when starting up on things that their firmware does not. (Things such as PLL startup etc). Whether these matter all that much is a good question.

The fact that you are not getting anything on the screen narrows it down to mostly the startup of the code base. I'll have a look through to see If i can spot any sticking problems soon.

rnelias, 2018-10-09:
as I've promised, here's the video showing the issue 

https://youtu.be/qVi4ksK9cgo

Ralim, 2018-10-10:
Hmm,
Does look like your device is hanging in the inital setup code. Let me poke around for a bit this weekend (or sooner)

FFY00, 2019-01-06:
Any updates? This is exactly what I am describing in #421

gerryk, 2019-01-29:
Seeing the same behaviour on a new iron... purchased in the last week.
Shipped with 2.18. DFU is 4.43. If I flash 2.06EN it just doesn't boot. Back to 2.18 it boots fine... heats up etc.

gerryk, 2019-01-29:
Tried 2.05 and it seems to work fine...

FFY00, 2019-01-29:
Can you open it and see the PCB version?

gerryk, 2019-01-31:
@FFY00 I can see v2.50 on the PCB beside the tip connector.

closed 2019-08-27 by Ralim