Ralim/IronOS #373: Ts100 no power after accidental over-volt

S4mFlynn, 2018-10-14:
Would have followed the template but this isn't a firmware problem, rather a hardware troubleshooting problem..  After connecting the iron to what **should have been** a 24v power supply, it turned on for a split second, then shut off with no smoke or hot components.  Naturally I took it apart, started testing components and going through the schematic you have posted on here.  After I figured out the RT7272B buck converter IC was bad, ordered 5 more and replaced it, but still no display on either USB or 18v power!  I did see a few diodes and found D3 was bad (reverse voltage protection diode?), is there anything else you guys know would fail past the reccomended voltages?  The MOSFET was never used since the iron wasn't heating up, so fingers crossed that survived..

Measured the output of that power supply at nearly 36 volts because it had terrible filtering, so that's what I get for using cheap equipment :/

Ralim, 2018-10-16:
Hi,
Sorry to hear this has happened 🤕 

First I would focus on USB power only to reduce some current paths. 
Remove the tip incase the mosfet is shorted so that it doesnt short your power supply.

IF the buck regulator failed, test if U5 has failed. If U2 is dead then the buck failed short and so the OLED is possibly dead, and if U5 is dead then the OLED and MCU are possibly fried. 
Depends how much time you want to spend on this thing, but happy to help 😄 

First I would power off USB, and try and validate voltage points on the circuit. I.e. what is the voltage at VIN of U2, the 3V9 point (output of L1), Vin of U5, Vout of U5 etc :) 

IF you can get it to the point of the oled coming on it will make diagnostics much easier :)

whitehoose, 2019-08-27:
Just to add.  Possibly similar to you, my TS80 died - screen just a few dots.  I contacted Banggood - described my problem and made a vid of the power on seq - they sent out a new package the next day by air - it landed only 10 days after.  That left me with a dead iron and good tip, case etc - so I contacted mini, told them what I'd done and asked if they'd sell me a motherboard and asked for prices with postage.

Guy said he would be interested to see where their protection had failed and asked me to send him my details and the mobo.  Day after my package landed he e-mailed me to say a replacement mobo was in the post.
5 days later I had 2 working ts80s with all the trimmings.

I was up front about my woopsie - unexpectedly both supports sorted my problem without pointing any fingers of blame (despite it being my cock-up).
It's maybe worth a go?

Couldn't ignore your comment "_it couldn't handle soldering large mechanical connections for things like rotary encoders at only 18v_".
Sounds like you need to consider a different, bigger Iron.  In my experience "only just" solder joints rarely make the grade.  many turn out to either be dry or brittle - either way they fail quite quickly - not good if you are relying on your rep to pay bills ... you need hot enough to make it quick not a slow cook. 

If you must have portable - until very recently I was still using lead acid and gas to make the big joins
The TS range is more a medium/light tool - and more specifically for hobby or last resort.  I evaluated the TS80 for work... we are now looking elsewhere - too many hoops to jump through.

Heavy, esp mission critical gear needs more robust tools.  Heat is physics, pretty isn't a substitute for thermal mass.

closed 2019-08-27 by Ralim