Ralim/IronOS #377: DFU 3.43 FW 2.18 - can't install your fw

fragolinux, 2018-10-17:
my ts100 arrived today, no way to change firmware

1)DFU 3.43 (Iron is on 2.18 Fw)
2)USB drive shows up as a 1.96MB drive.

keep pressed button near tip, it shows as drive F:
i put TS100_IT.hex in it, it disconnects, i reopen it and there's a TS100.err file, which i cannot open nor copy on pc... if i keep putting the hex file there, the TS100.err files duplicate and triplicate, all with the same name, which is weird...
tried both latest 2 firmwares version actually on releases page

dhiltonp, 2018-10-17:
The build up of garbage .err files is normal behavior when flashes fail.

This probably isn't related to the firmware you've selected.

Most likely there's something interfering with the magic incantation (it really is pretty sensitive).

Double check that you're doing exactly this: https://github.com/Ralim/ts100/wiki/Upgrading-Firmware#windows

A 3rd-party utility may be interfering - maybe a virus scanner, etc.

fragolinux, 2018-10-17:
you're the boss! It was, indeed, Teracopy (single R, not double as in the link you suggested)... there was no mention to it in opening page where the info are presented: too, better add there, too...
and i just did "right click - unlock file" on the just downloaded file, as an addon precaution... thanks!

closed 2018-10-17 by fragolinux

fragolinux, 2018-10-18:
> @fragolinux Glad this worked for you in the end, their DFU code is rather picky to deal with sadly :(
> 
add teracopy info to main page, it's missing there: https://github.com/Ralim/ts100

EazyFPV, 2020-02-13:
I cant get my TS100 to Update I see DFU 3.45 originally was on 2.18. I see RDY when I copy the files over but nothing happens when I plug my TS100 in