Ralim/IronOS #395: BAD TIP at temps >400C

dhiltonp, 2018-10-29:
I'm getting bad tip when I cross 400C or so.

Can you verify that it works for you?

https://github.com/Ralim/ts100/blob/master/workspace/TS100/src/main.cpp#L540

Ralim, 2018-10-29:
Ah, with the new tip calibrations, some tips cant get up to 450C before they go out of range of the unit.
Because the new tip profiles are more accurate some of the models can only hit around 430C before they are out of range of the ADC reading back the temperature.

Any chance you have a tip thermocouple you could verify temperature accuracy with ?

I'll merge in your update and then test this tonight either way :)

May need to limit the max selectable temperature depending on the tip.

dhiltonp, 2018-10-30:
I think there's a little more to it than that, I could easily replicate it going to 400C, but I first noticed it when lowering the temperature from 320C to 300C.

I don't have a tip thermocouple. It could be useful...

Ralim, 2018-11-03:
I don't think this is your tip, reproducible on two of mine.
Looking into this further as I have time.

Ralim, 2018-11-03:
I'm going to have to look into this tomorrow.
On all of my ts100 units here, I get 1 of two failure modes whenever I try and solder now.
Either:
a) On power up the unit is reading a high temperature >400C and is constantly drawing max power
or
b) The temperature regulates well but the OLED display locks up, and ignores all user input.

When compiled using gcc command line, with full LTO the binary is 3.5k smaller, and instead seems to trip the bad tip all the time whenever the output is on.

I have no idea what Atollic is doing in the first case, For the latter case I think its most likely PWM timings that I'll look into tomorrow.

dhiltonp, 2018-11-05:
I built with sw4stm32 and didn't see either a or b.

When I set the iron to heat on power-on, I did see an undervoltage warning not present in older versions.

Ralim, 2018-11-06:
Hmm interesting, Atollic is using a newer compiler than sw4stm32 so not sure what the compiler is changing.

Damn, need to update the code to pre-fill the voltage check buffer.

Ralim, 2018-11-09:
I've pushed a small fix to master that helps with the startup voltage sensing (not ideal, but works)

Also after this change my ts100 is working really well.

However, trying to get this working on the TS80 now. It oscillates with a period of about 2 Hz with the current settings. What is your suggested tuning method here ?

Seven5555, 2019-01-01:
I'm not if this where this goes but it seems like the right place and thing to do. My TS100 constantly getting this "Bad Tip" message, since installing the new firmware over the stock firmware yesterday. I just got this a little over a week ago so it should not have bad tip. 

I do want to thank you for creating this very cool firmware with all these much needed extra settings, I like it a lot for sure.

Dean

Ralim, 2019-01-01:
@Seven5555 This is #422