Ralim/IronOS #401: DC_LOW and half screen goes blank when heating up

nmaggioni, 2018-11-08:
* **I'm submitting a ...**
  - [X] Bug report
  - [ ] Feature request
  - [ ] Translation

* **What is the current behavior?**
Master (as of 18a8461) is broken: since commit 73f0748, whenever I start the heating, the upper half of the screen goes blank and `DC_LOW` appears.

* **What is the expected behavior?**
The iron should heat up normally and the screen should stay fully functional.

***Steps to reproduce:***

1. Flash a firmware build since (including) commit 73f0748.
2. Connect the iron to a power source.
3. Start the heating.

***Video of problem if hard to reproduce:***

https://i.imgur.com/Tnv57ub.gifv

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.06
  - PCB Version: 1
  - Power Supply (Voltage and Current Rating) : 12V, 2.25A

* **Other information**
Here's a build of the master branch (as of 18a8461) with commit 73f0748 reverted - my iron now does not show the aforementioned problem anymore: [TS100_EN.zip](https://github.com/Ralim/ts100/files/2562405/TS100_EN.zip)

Ralim, 2018-11-10:
Hi, could you give the master branch another go. Hopefully that will fix the DC LOW warning.
The screen going half blank is a much more annoying issue to troubleshoot (something is overwriting the OLED buffer somewhere)

nmaggioni, 2018-11-11:
In the latest master both problems now seem fixed :+1:

nmaggioni, 2018-11-11:
I was too quick, the next time I powered the iron on the problem reappeared and now presents every time I use it.

nmaggioni, 2018-11-19:
Sorry it took me this long! I tried flashing master as of today and resetting to stock, but it does nothing to fix the issue. Is there any other debug data that I could provide?

I've already isolated the issue in the commit mentioned above, could it be that [the size of the struct's buffer](https://github.com/Ralim/ts100/commit/73f07489fee90d35169dd6da2a7f2c9b685358c4#diff-16ebd4ec28c7fd3c1d6919183919a077R16) was changed (`75` -> `3.5*32`)?

Ralim, 2018-11-19:
No rush :)

The problem I have with that is, that the buffer is used for historical I terms of the PID and as such having it larger would only be backing off the initial power used when heating. I'll have another look for maths issues but it's unlikely that the buffer change caused this.

It's far more likely that your iron is having issues with the updated PWM settings introduced in earlier commits and activated in that commit.

I can only cause DC low on my irons when I use power supplies that are weak. The newer code that the commit activates improves the tip response and smooths out the heating curve. But it does mean you power supply must meet spec (2A with low voltage sag).

As your running 12V, I would be most suspect of your cabling having a voltage drop that pulls the input below the 10V cutoff when under load. Is it possible for you to measure the voltage inline to look for sagging under to load?

nmaggioni, 2018-11-19:
I'll try to make a harness to test that, even though the 220V supply is rated for 12V @ 2.5A. Or, better yet, I'll hook it to a proper lab PSU and see if that holds up.

Ralim, 2018-11-19:
That would be wonderful.

I've had supplies before that sag under loads fairly badly which is disappointing. That said I'm thinking of just removing the lower limit in DC mode for these issues.

nmaggioni, 2018-11-21:
I tested it today with the latest build and the said lab PSU and the behavior didn't change, unfortunately. As soon as I turn the heating on, DC_LOW appears and the upper half of the screen goes blank (black) until I reboot the iron.

closed 2019-08-27 by Ralim

andypiper, 2020-12-02:
I recognise that this is an older and closed issue, but this (the DC LOW problem rather than the half screen one) has happened to me on a TS100 today. I've tried a couple of different USB cables and power supplies, both on 2.11 (which was the update I did this week), and also after a downgrade to 2.05 which appears to potentially have been a last known good based on this issue. Do we know what the root cause was - am I not using an appropriate supply? To be honest, I haven't used my iron for a year and a bit, but apart from running a firmware update, nothing else had changed that I could tell.

Ralim, 2020-12-02:
If it's just DC LOW it could be power supply sagging;
Would it be possible for you to test a secondary supply by any chance ?

Also going into the debug menu (hold the rear button on the home screen) and using the front button to cycle through and check what vin reads as would be good to sanity check