Ralim/IronOS #408: Cannot get logo to work.

ShotgunBurger, 2018-11-17:
- [x] Bug report

* **What is the current behavior?**
After I update with FW TS100_EN.hex released Aug 2nd my iron works great but after dragging in my hex file with the logo, the iron wont turn on at all, to fix it I re-flash with the firmware and it works fine.

Any ideas?

Ralim, 2018-11-17:
The logo format changed at one point.
Regenerate your logo for with the latest software and try that again first. :)

ShotgunBurger, 2018-11-18:
I'm creating the logo fresh each time I try  using the logo creator but it still doesn't work. I might just have to run the iron without any logo.

Ralim, 2018-11-18:
Can you upload your logo HEX file here and I can test (it will need to be zipped)
Also can you provide the information you cut out of the issue, at the end (unit details, firmware details)

ShotgunBurger, 2018-11-19:
Here is the logo.
[logo.zip](https://github.com/Ralim/ts100/files/2594683/logo.zip)

Here is the firmware I'm using.
[TS100_EN.zip](https://github.com/Ralim/ts100/files/2594687/TS100_EN.zip)

And the DFU is 3.43, that's all the info I have sorry.

Ralim, 2018-11-23:
Hi, 
Looking at your logo file, You are using the old tool to convert the logo.
There is an updated tool in the releases section if you want an `exe` or you can use the python tool (which does often do a better job).

ShotgunBurger, 2018-11-24:
Awesome, cheers.

closed 2018-11-26 by Ralim