Ralim/IronOS #409: Russian in v2.05.01 is not working

AlexGyver, 2018-11-20:
Hello! Firmware v2.05.01 Russian is lack of fonts! Only icons

Reydan46, 2018-12-11:
To work, you need to recompile from source.
I recompiled in Win 10. [Here](https://www.dropbox.com/s/xbgp77pjamgr9rk/TS100_RU.hex?dl=0) is the working version.

artemirq, 2018-12-11:
@AlexGyver Как мне кажется, оригинальная прошивка все же стабильнее.Так же нагревает жало быстрее

dhiltonp, 2018-12-12:
@Reydan46 : so it works on the current branch?

artemirq, 2018-12-12:
@AlexGyver Да, это удобно. Но вручную повысить температуру - парочка нажатий. Делов-то. Турбо режим не такая уж и киллерфича. В остальном родная прошивка такая-же или, скорее всего, даже лучше. Анимация, например, симпатичнее в родной. Настройки можно менять на лету в ласт версии
И еще, раз мы тут, хотел-бы поблагодарить за твой видео-контент и за сайт. Созидаешь не за деньги, а от души, как истинный творец. Все офигенно, продолжай в том же духе. Успехов 👍

AlexGyver, 2018-12-12:
> @AlexGyver Да, это удобно. Но вручную повысить температуру - парочка нажатий. Делов-то. Турбо режим не такая уж и киллерфича. В остальном родная прошивка такая-же или, скорее всего, даже лучше. [...]

Спасибо) я лично без турбо режима уже пайку не представляю. А насчёт не за деньги, не надо делать поспешные выводы :)

artemirq, 2018-12-12:
@AlexGyver Я про то, что твои видео наполнены идеями твоего вдохновленного разума. Есть блогеры, у которых идея видео основана на том, чтобы что-то разрекламировать и распиарить. То есть они тупо на максимум извлекают прибыль, а не хотят сделать что-то качественное и полезное для других, чем можно было-бы гордится

AlexGyver, 2018-12-12:
> @AlexGyver Я про то, что твои видео наполнены идеями твоего вдохновленного разума. Есть блогеры, у которых идея видео основана на том, чтобы что-то разрекламировать и распиарить. То есть они тупо на [...]

Здесь да, с душой)

closed 2019-08-27 by Ralim