Ralim/IronOS #412: Update installation instruction in case a .ERR file is created -> use command line to copy file

ventilator, 2018-11-27:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ ] Bug report
  - [x] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**

Improve installation instructions

* **What is the current behavior?**

Could not install firmware update on a Windows 10 System with Teracopy installed.
Installation instructions suggest to disable Teracopy during .hex file copy. This did not help.
But copying the firmware with command line "copy" command it worked.

* **What is the expected behavior?**
Add a sentence to the installation instructions, if there is a .ERR file created, try using the command line to copy the file to the soldering iron

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. 
2. 
3. 

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**
user experience improvement

* **What are you running:**
Win 10 x64

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.x
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :

* **Other information** 

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

19Gerhard85, 2018-11-27:
There is already a warning in the [Wiki](https://github.com/Ralim/ts100/wiki/Upgrading-Firmware).

> Using Teracopy, or other explorer replacements will make it fail.

Ralim, 2018-11-28:
@ventilator If you feel that this would be helpful to have , feel free to submit a pull request to change the wording to something that would have helped you. (you can edit it in the web-browser to make a pull request easily)

ventilator, 2018-11-28:
> 
> 
> @ventilator If you feel that this would be helpful to have , feel free to submit a pull request to change the wording to something that would have helped you. (you can edit it in the web-browser to make a pull request easily)

Thanks, again I learned something :-)

Courty40, 2020-02-06:
i'm getting this type of error with the latest version of Windows 10 where it seems to do a pre-copy or the like resulting in an ERR file.
Going to the command line (run/CMD) and typing "copy ts100.hex d:/" (if D: is your TS100) and then unplugging the iron makes it work every time where as using the file manager to drag and drop was resulting in 100% fails.

discip, 2022-04-23:
@ventilator
Please close this issue. 😊

thanks

closed 2022-04-23 by ventilator