Ralim/IronOS #414: Binary releases no longer available

lorenz, 2018-12-01:
I run DAPBoot as DFU bootloader. When switching to per-language releases it looks like the binary releases were no longer built. 

* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Workaround** 

```
arm-none-eabi-objcopy -I ihex --output-target=binary TS100_EN.hex TS100_EN.bin
```

dogtopus, 2018-12-18:
I agree it is pretty annoying for DAPBoot users (In fact I am also using DAPBoot on my TS100). However the problem is that the release page is already pretty crowded due to the number of multi-language firmwares. It would probably be worse if binary versions are also provided. For this reason I always build my own binary using the Makefile instead of relying solely on the release.

JohnEdwa, 2018-12-18:
Seeing as all of the current .HEX releases combined are only around 2.5MB, grouping all the non-english language releases to a single .zip file could clean the release page while really not making that much of a difference. Although the filename would be pretty funny with all the 20 or so language codes back to back.

So:

* TS100_EN.HEX - 130KB
* TS100_EN.BIN - 130KB
* TS100_BG_BR_CZ_DE  ...  TR_UA.ZIP - ~5MB

lorenz, 2018-12-18:
I also used to build my own but none of mine work, probably due to some undefined behavior and GCC 8 behaving differently. So I'm kind of stuck using the officially built ones.

closed 2019-08-27 by Ralim