Ralim/IronOS #416: Cooldown blinking on Detailed idle

michalszymura, 2018-12-04:
* **I'm submitting a ...**
  - [x ] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
Tip temperature is not blinking / No visual indication of hot iron

* **What is the expected behavior?**
Tip temperature blinking (previous version i run was 2.0 and it had additional screen with a warning)

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. Activate:  Detailed idle screen
2. Activate: Cooldown blink
3. Start soldering
4. Stop soldering
5. Temperature is not blinking while cooling down

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**
SAFETY!

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.05.01
  - PCB Version: 1
  - Power Supply (Voltage and Current Rating) : 24V 5A

* **Other information** 
Similar issue to #289 but on Detailed idle screen
If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Strat00s, 2019-02-03:
Can report the same behavior on the latest FW (2.06). Detailed idle screen won't blink, but the simple one does (and way faster than on the version I was before, which is about a year old now).

Ralim, 2019-02-06:
@Strat00s Funny how it's not tagged as fixed and its not fixed :scream: 

Not sure if I will add this or not since it serves little purpose on the detailed screen as the point of the detailed screen is for the user to be capable of reading? Intent is for most users to use the normal idle screen unless the want to know the nitty gritty.

lepicfoto, 2019-04-30:
Yep the latest software and exactly the same problem, it's a shame it was a great good feature. Jesper malling

Ralim, 2019-05-01:
@lepicfoto

Can I confirm why you want this feature on the detailed idle screen ? The intent of that screen is for users who want to read the information. Flashing the small text for the temperature isn't overly eye catching for this purpose, so i felt it was actually safer to go without so that users of the detailed screen would read if unsure, rather than miss the small flash?

jj-uk, 2019-06-11:
At the moment, when the button is pressed and held, the iron starts to cool. The flashing screen indicated that the iron was cooling down, it stopped flashing when it was below a safe temperature.

Now, the screen shows that the iron is cooling but it does not flash, meaning I need to pick it up and check the temperature when waiting for it to cool enough to pack away. 

What would be good is if it flashed COOLING, until it reached 30C, then the screen simply shows 
SAFE - TEMP 30C 
in large letters, and nothing else.

The screen at the moment turns it's self off after a while. This leaves me guessing if the battery went flat and the iron is still hot, or the iron turned off due to no movement. I'm never reallty sure if its cool or not anymore. Please add the flashing cooling screen back in, and a COOL - SAFE screen that is ALWAYS shown if the iron is safe (never turn off the screen).

jj-uk, 2019-06-11:
At the moment, when the button is pressed and held, the iron starts to cool. The flashing screen indicated that the iron was cooling down, it stopped flashing when it was below a safe temperature.

Now, the screen shows that the iron is cooling but it does not flash, meaning I need to pick it up and check the temperature when waiting for it to cool enough to pack away, sometimes it's just in standby, so it heats up again, not safe because sometimes i forget to tell it to cool down, and forget that i didn't tell it to cool down.

What would be good is if it flashed COOLING, until it reached 30C, then the screen simply shows
SAFE - TEMP 30C
in large letters, and nothing else.

The screen at the moment turns it's self off after a while. This leaves me guessing if the battery went flat and the iron is still hot, or the iron turned off due to no movement. I'm never reallty sure if its cool or not anymore. Please add the flashing cooling screen back in, and a COOL - SAFE screen that is ALWAYS shown if the iron is safe (never turn off the screen).

whitehoose, 2019-06-11:
If you travel, best idea is to have either an aluminium or silicone tube the same ID as the OD of the cone at the base of the iron's shaft. Ali cools quicker and silicone cushions better - so look at your arrangements to see which is best.

The sensor is at the end of the tip - so it's possible (I'd have to agree unlikely) to still have a hotspot elsewhere whatever the display tells you.  I routinely sheathe mine and pack it while hot - Giving copper a slow cool isn't a bad thing - stops the tip getting brittle - this goes for the conductors as well - so it's a bit of a win win situation and could extend the life of your kit
I've done this all my working life, a big copper slug takes a long time to cool - not convenient to wait - and not a good idea to quench.  Smaller Irons don't have the mass - but the old principles still apply

Ralim, 2019-08-27:
@michalszymura 

To avoid packing in another string and more mess for translation, what are your thoughts on if i combined this with #476 , and had an option setup so that for the idle screen you could pick:

* Normal
* Always inverted
* Flashing inverted when iron is hot

This would allow for colour on the screen to invert when the tip is hot as a way to catch attention, and also solve #476 at the same time?

michalszymura, 2019-09-05:
@Ralim yeah sure, i'm fine with it.

closed 2019-09-05 by michalszymura