Ralim/IronOS #417: LiFePo4 element

tweakerm, 2018-12-09:
LiFePo4 battery is incorrect voltage cutoff.
It is necessary to tune the selection of the type of batteries, as when working with LiFePo4 batteries there is a different cut-off voltage per bank than LiPo

* **I'm submitting a ...**
  - [ x] Bug report
  - [ x] Feature request

* **Do you want to request a *feature* or report a *bug*?**

* **What is the current behavior?**

* **What is the expected behavior?**

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. 
2. 
3. 

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.x
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :

* **Other information** 

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Ralim, 2018-12-10:
_That Voltage limit of ~10V is there for a reason._
At that voltage the iron is not overly heating up above ambient well as you are only drawing around 10W or so. Please look into a higher cell count battery rather than complain about this limit. 3S LiFePo4 is not a recommended supply for using a TS100. 10V is used because at about 8 volts quite a few tips fail to heat up at all above a cool environment, and they cannot solder.

tweakerm, 2018-12-10:
I understand that, but when I set the lower limit to 9 volts, my copy perfectly solders with these elements.

closed 2019-12-29 by Ralim