Ralim/IronOS #419: Question / Feature request: Lower min temp?

michael-simon-eu, 2018-12-23:
* **I'm submitting a ...**
  - [ x ] Feature request

* **Do you want to request a *feature* or report a *bug*?**

I'm not sure if this is the right place for this question, but i don't where else to ask.

Is there a possibility to change the minimum temperature to somewhere lets say below 70°C?  

* **What is the motivation / use case for changing the behavior?**

Use case would be for me in creative work manly in the kitchen (design work with chocolate) but also for design work on candles.

whitehoose, 2018-12-24:
Interesting!  I've sometimes thought of a bit of pyrography, hadn't considered the other, cooler end of that spectrum, but I have a tool that cuts plastic bottles into a thin filament then use the "string" to bind joints.  Once bound you heat the join and (if you get the temp right) it tightens to make an incredibly strong join or tough protective sleeve.
I can see the potential.

Ralim, 2018-12-26:
Current minimum temperature is 50 deg C / 120 Americans ?

michael-simon-eu, 2019-01-14:
You are right, it is. That's grate! 
(I was irritate here because every spec says 100 - 400 °C )

closed 2019-08-27 by Ralim