Ralim/IronOS #421: TS100 No Power After Firmware Update

TheGoldenAir, 2018-12-28:
After updating TS100 firmware to TS100_EN.Hex will not power on using AC adapter or Lipo battery. I have downloaded the update and installed multiple times, but the issue still presents. I am waiting for the file to convert to .RDY before disconnecting from computer. I can get back into dfu mode and when I do, the drive shows empty.

turbosi92, 2019-01-01:
I have this EXACT problem.  Had a working TS100 that was on a really old firmware.  Decided to upgrade and this happened.  Any fix?

TheGoldenAir, 2019-01-01:
No luck on my end, and I cannot find any input anywhere. It is not bricked because it still loads in DFU. Very frustrating.

turbosi92, 2019-01-01:
Right.  Dammit!  Did you update your firmware on windows?

TheGoldenAir, 2019-01-01:
Yes, I did.

Ralim, 2019-01-01:
Which firmwares have you tried?
Which was the last known good firmware for you ?
What hardware revision do you have?
Are you doing the double copy until you get RDY extension?
Did you use Logos (if so have you applied them again?)

turbosi92, 2019-01-01:
I tried the Latest.
From December of last year I think?
Not sure but it was purchased in November (Black friday ) 2017
Double Copy?  Explanation?  Link?
No logos in my case.

It won't power up via USB either.

Ralim, 2019-01-01:
Latest being 2.05 or 2.06-RC1?
Most likely a version 1 PCB then.
Copy the file to the usb drive twice before ejecting (ie once, wait for a bit for it to re-appear, then copy again, wait again, then unplug).

No logos is good as it makes this easier.

TheGoldenAir, 2019-01-01:
I used 2.06 (the first EN file on the release page). 
I tried copying the download to the drive multiple times before disconnecting. 
Not sure of the original firmware on the iron, but it was purchased new in Feb. 2018.

fullmetal03, 2019-01-01:
have you guys experiencing issue that usb plugged but it didnt show up on autoplay however i see my DFU was 3.45 mostly people got 3.42 so I wonder whats it does with 3.45??

Ralim, 2019-01-02:
@fullmetal03 Can you confirm if you have DFU 3.43 or 3.45 or 3.42?
3.45 is normal on TS80, 
3.43 is normal on TS100
3.42 I havent seen

Ralim, 2019-01-02:
@turbosi92 If you go back to the 2.05 does it work for you ?

turbosi92, 2019-01-02:
> @turbosi92 If you go back to the 2.05 does it work for you ?

Unfortunately no...seems its dead.  :(  If you have something else you want me to try, let me know.  I leave tomorrow for a week so won't be around to test anything.

fullmetal03, 2019-01-02:
> 
> 
> @fullmetal03 Can you confirm if you have DFU 3.43 or 3.45 or 3.42?
> 3.45 is normal on TS80,
> 3.43 is normal on TS100
> 3.42 I havent seen

im on TS100 and the DFU is 3.45 but it wont autoplay as appear device.. is it something wrong?

FFY00, 2019-01-04:
Hello, I am experiencing the same issue too! I just got my iron and now it's broken.

After flashing `v2.06-RC2` I tried flashing `v2.05.01` and the original firmware. No luck.

FFY00, 2019-01-04:
Hey, so apparently flashing [ts100tris](https://github.com/joric/ts100tris/releases) (based on `v2.17`) worked on my mouse.

turbosi92, 2019-01-05:
FFY00, You only flashed the app.Hex file to fix it?   I'll try when I get home on Tuesday

FFY00, 2019-01-05:
Yep, only `APP.hex`. It seems like all other hex files are bad, even the official ones. This is super weird.

Heppie, 2019-01-05:
I had the exact same issues. My issues resolved after uploading the new hex files via a USB 2.0 port rather than a USB 3.0 port on my PC. Its worth a shot!

FFY00, 2019-01-06:
For some reason I was able to flash the official firmware. This one still doesn't work, it flashes but I get a blank screen. @Ralim would the flash log help?

turbosi92, 2019-01-09:
> Apparently flashing [ts100tris](https://github.com/joric/ts100tris/releases) (based on `v2.17`) worked on my iron. This is weird because I tried flashing both `v2.18` and `v2.17`. Could this issue be happening because of the build environment? @Ralim how are you building the binaries?

Unfortunately this didn't work for me.  I ordered a new TS100.  :(

turbosi92, 2019-01-09:
Where can I find the original firmware to try that?

sefer2, 2019-01-29:
I have DFU 3.43 on my TS100 tried the 2.06rc3 (it appeared to install correctly and the .RDY file appeared) however the screen wouldn't even light up afterwards.
Reinstalled 2.06rc2 and it came back to life.

rozwell, 2019-01-29:
A little of my input to this subject, regarding **v2.06-RC3** :)
### Which firmwares have you tried?
1. **Original v2.18** => works
2. **Ralim v2.05.01** => works
3. **Ralim v2.06-RC3** => doesn't boot, only DFU works, connecting via USB results in unknown device

I can switch between above 3 without issues, RC3 just doesn't boot.
### Which was the last known good firmware for you ?
Like above, but I didn't compile anything myself yet, like RC1.
### What hardware revision do you have?
```
v2.05 PCB2
DFU: 3.43
```
### Are you doing the double copy until you get RDY extension?
Getting .RDY extension at first try
Tried double copy and ended up with two .RDY files, RC3 still doesn't boot.
Used USB 2.0 and 3.0 with same results.
### Did you use Logos (if so have you applied them again?)
Nope
### When testing just use the USB 5V to power up to narrow down the search space for issues.
Connecting via USB results in unknown device on RC3

vasyapupkinodin, 2019-01-30:
Hello. I have ts-100 dfu: 3.45 v2.18. On version v2.06 RC3 (TS100_RU.Hex) I have a black screen. On the rest up to v2.04 psb2 I have no text. And only on version 2.04 psb2 works for me. Power supply 19v 2.5a.

FFY00, 2019-01-30:
My TS100 is still dead on RC4. What compiler did you use? Perhaps IAR could fix this.

Ralim, 2019-01-30:
@vasyapupkinodin Can you please test the EN version as well, to narrow down Russian vs English. Russian is by far the most annoying language for support :/ (I dont dislike you guys, just annoying it keeps breaking!)

@FFY00 If you are willing to provide a paid for license for IAR then I'm all ears, but unless you are willing to give one away (Legally) its not really an option, Also building under linux for IAR with C++ is often anything but fun. As per notes elsewhere this is all built using GCC for arm.

The official firmware is not the best at recovering from flash contents changing from what it expects, depending whats in yours flashing an odd sequence gets it into a state it likes. 
They claim there are no clone units, but there appear to be a very small percentage of users that have units where the OLED wont initalize properly according to spec (often it just doesnt ACK any I2C messages its sent). Their firmware implements the I2C using bit-banging (for slow speend naturally) where as I use the hardware peripheral to offload CPU time.

FFY00, 2019-01-30:
@Ralim, I currently have a evaluation license so if you could so if you could send me a IAR project file, I could see if that's the issue. If that's indeed the issue, I could see if my college currently has any license and if not, I could possibly request one. Unfortunately I wouldn't be able to give you the license but I would be able to build the project when needed.

blizardnl, 2019-02-18:
I had the same problem. Solved!  Press the button nearly to iron (same procedure as attaching) while disconnecting from PC. Now the file does not disappear anymore. Hexfile used 3.45

SputnikSeven, 2019-03-28:
I had this same issue, and what it turned out to be was a faulty data cable. When I used a different cable the copy worked perfectly. I would also try the copy on a different computer if possible.

siggey, 2019-04-13:
DFU 3.45,  Press the button nearly to iron (same procedure as attaching) while disconnecting from PC
IS THE ONLY WAY.
so the procedure is:
1- connect pressing the A button
2- leaving the A button pressed disconnect
3- reconnect pressing the A button

TilTisback, 2019-06-09:
> 
> 
> A little of my input to this subject, regarding **v2.06-RC3** :)
> ### Which firmwares have you tried?
> 
>     1. **Original v2.18** => works
> 
>     2. **Ralim v2.05.01** => works
> 
>     3. **Ralim v2.06-RC3** => doesn't boot, only DFU works, connecting via USB results in unknown device
> 
> 
> I can switch between above 3 without issues, RC3 just doesn't boot, used EN version.
> ### Which was the last known good firmware for you ?
> 
> Like above, but I didn't compile anything myself yet, like RC1.
> ### What hardware revision do you have?
> 
> ```
> v2.05 PCB2
> DFU: 3.43
> ```
> 
> ### Are you doing the double copy until you get RDY extension?
> 
> Getting .RDY extension at first try
> Tried double copy and ended up with two .RDY files, RC3 still doesn't boot.
> Used USB 2.0 and 3.0 with same results.
> ### Did you use Logos (if so have you applied them again?)
> 
> Nope
> ### When testing just use the USB 5V to power up to narrow down the search space for issues.
> 
> Connecting via USB results in unknown device on RC3

I can confirm this 

Exept i have DFU 3.45

colbycarp325, 2019-07-09:
From reading most of these im assuming that my ts100 is just broken and there is no way to fix this?

Ralim, 2019-07-10:
If you can boot into DFU then in theory you can load the firmware.
If the DFU is rejecting the firmware images you could try replacing the miniware one with the USB-DFU one.

If the DFU accepts the firmware but the iron doesn't boot, make sure to try the English version.
I have a new build coming up this week that fixes a stack of the language issues.

After that I will be looking into this more, but it's hard to debug without reproducibility.

If you have an iron that doesn't boot, it would be helpful to get a flash dump off it via the SWD link.

dzalf, 2019-09-01:
Hello everyone!

Disclaimer: I am re-posting this comment here because is related to issue #482 where the unit wouldn't turn on after trying to upgrade the firmware and this is how I solved it.

"Guys.

 I just want to confirm that the USB cable is **REALLY** important.

Symptom: The TS100 bricked after several intents for copying the file. Unit bricked after that. Dead.

Here's how my case developed with the _faulty_ cable:

1. I entered into DFU (3.45) pressing button _A_. Windows 10 recognised it as a 1.95 MB disc drive
2. I dragged and dropped the hex file into it. Here's where the issue happened:

The TS100 disconnected itself from Windows **before** the file transfer completed.

I tried different ports (USB 2.0 and USB 3.0 back and front) without success.

Strangely this USB cable is my favourite one. I do Arduino/Teensy/PIC programming on a daily basis and the cable has ALWAYS worked like a charm.

SOLUTION:

I used another _crappy_ cable that I found on a drawer. I connected the TS100 to a back USB 3.0 port and _voila_. The file was successfully transferred and the unit came back to life. Happiness.

I must point out that the cable that made the trick was **shorter** (about 15 cm) vs the other _faulty_ cable that is ~1 m. I insist: the first cable is my daily driver for programming Microcontrollers and it still works. i just tested it with a fast serial communication with no problems.

Moral of the story: never trust your _trusted_ cables.

Thank you @Ralim for this amazing firmware!

I'm inviting you a :beer: for sure ;) (for real. I'll go right away to the donation page)

dzalf

😎 "

jeckel2788, 2019-09-04:
i am having the same problem that alot of people are stuck in dfu after trying to update. i tried alot of different things. maby if i used a diff program/driver than windows it may help. getting no disk and errors alot. alot of people are bricking there unit , you can go back to the stock firmware, which i did until this gets worked out. here its a link. download the zip file.

http://www.minidso.com/forum.php?mod=viewthread&tid=892

 @jeckel2788

F1l3N0tF0und, 2019-12-26:
So. I got a brand new TS100 for Christmas this year (yay ^^) and i had the same problem where the Iron would just disconnect when attempting to write the FW file (Ralim 2.05.01) and reconnect as a USB Drive without a name. After that I only was able to bring it back to life by flashing the official firmware by Miniware.
It is an original Miniware Iron with DFU V3.45

I tried multiple USB Cables and multiple ports on my PC (Win10) and I also tried it with and without TeraCopy (fore those who are unfamilliar with that Program, it works just like copying files with the explorer but is does a Cheksum verification afterwards). What ended up working for me was using another PC (also Win10), using a USB 2.0 Port which supports higher Current over USB (more than 500mA) and using the Explorer to Copy the FW File.

Hope this helps :)

Ralim, 2019-12-26:
Good Christmas present :tada: :gift: 
Hope it works well for you!.

You _definitely_ have to use windows explorer to copy the file.
Their software makes assumptions about the order the file is copied over the USB link, which causes issues if you use _any_ software that likes to interfere with that process (Antivirus, TeraCopy, AntiMalware etc)

closed 2019-12-29 by Ralim

YulawOne, 2020-06-08:
Just had the same issue, when tried to update firmware using USB 3.0 with DFU v3.45, after it started googling and have found this issue and tried USB 2.0 and it worked... Could you please update installation wiki with information regards USB 3.0 issues?

Ralim, 2020-06-08:
@YulawOne I did add a note to the flashing info page to reccomend trying different ports. In this case its less about 3.0 vs 2.0, and more about just a different controller chip thats more tolerant.

> Secondly, try other USB ports. Often different USB controllers will interact with the units differently due to design quirks in the miniware design.

YulawOne, 2020-06-24:
Sorry for missing this info, I was looking into https://github.com/Ralim/ts100/wiki/Upgrading-Firmware, which differs from https://github.com/Ralim/ts100/blob/master/upgrading.md

colbycarp325, 2020-06-24:
I had the same problem with it not working after the update, all i did was use a different computer to update it again and it worked after that.

Ralim, 2020-06-25:
@YulawOne  they are different as the one in git is meant to be a slightly more concise start point, and then the wiki can be community updated to add more detailed information

PumaFPV, 2020-09-08:
Hey I just bought a TS100, DFU 3.45 ver 2.18 stm pcb v2.51A power pcb v2.51B. I tried flashing you firmware but when i do, oled stays black and the iron doesnt consume any amp... I flashed the 17 Sep 2017 version. 
Only firmware that seems to work is one from minidso v2.18 as it is called.
Does anyone has a solution?
Thanks

Ralim, 2020-09-11:
@PumaFPV 

>17 Sep 2017 
Are you _sure_ you flashed this old version (1.17.1) ?
That is _very very_ out of date, latest release is 2.12 beta or 2.11.

3.45 and 3.43 DFU have _different_ problems; not strictly better or worse.

If your getting the .RDY file then it most likely did flash. Can you please test the latest release (2.12 beta, TS100_EN.hex) to validate why this is occuring?

PumaFPV, 2020-09-11:
Thanks now it works ! I cant even remember why i tried those old versions... 
Anyway, now i have an almsot perfect ts100. Did you see the oscilloscope mod? I'll give it a try since this would be the perfect compact tool :)

Ralim, 2020-09-11:
@PumaFPV 

Indeed I have seen a few mods. I'm a big fan of keeping tools for their intended purpose so that the protection is well designed 🤣

lampros2213, 2020-10-01:
hello

im trying to upgrade the firmware but i have a problem
i have a ts100 im entering dfu mode (dfu 3.43)

i copy and paste the .hex file and after the reset it says .rdy but whenni plug it on the power the screnn stays black and never turn on can anyone help me?

if i they the official firmware the soldering iron is back to normal and working fine

paulfertser, 2020-10-01:
Hello,

On Thu, Oct 01, 2020 at 07:37:20AM -0700, lampros2213 wrote:
> im trying to upgrade the firmware but i have a problem
> i have a ts100 im entering dfu mode (dfu 3.43)
> 
> i copy and paste the .hex file and after the reset it says .rdy but whenni plug
> it on the power the screnn stays black and never turn on can anyone
> help me?

Can you please provide more details? What specific versions of Ralim's
firmware have you tried? Is your soldering iron original or clone,
where was it bought from?

-- 
Be free, use free (http://www.gnu.org/philosophy/free-sw.html) software!
mailto:fercerpav@gmail.com

paulfertser, 2020-10-01:
> sry i dont have swd debugger or experience :S

Can you probably give it temporarily to someone who has? And who can
probably provide ssh access to a machine with debugger and this iron
connected? It's probably something trivial but needs a developer to
fix, and the problem manifestation (no display output at all) provides
no possible clue for debugging.

Have you tried flashing Ralim's firmware and then connecting power
supply and pressing the button that's close to the tip (not the
display). Even if display doesn't work that should still lead to the
iron heating up.

lampros2213, 2020-10-01:
if i put the new firmware for example the 2.12 the screen not working but the iron is getting hot
and if put power and after press the button closer to the tip its getting back to dfu
unfortunately i dont have any friend that have SWD debugger 🈂️ 
i think i will stay with the official firmware :S

paulfertser, 2020-10-07:
@lampros2213 and the others: could you please tell if you're seeing similar results to what's visible in #666 video? Please try
https://github.com/Ralim/ts100/suites/1300796821/artifacts/20410913
that was auto-built from my test "github pull request" and see if the
behaviour is any different.

lampros2213, 2020-10-07:
@paulfertser 
yes its doing the same thing i try the hex file that is attached in the comments but it didnt fix it its doing the same thing 👍 
no infos at the screen but if i press the button near the tip the iron starting getting hot like it was suposed to be but nothing on screen :S

paulfertser, 2020-10-07:
Thanks for testing!

> yes its doing the same thing i try the hex file that is attached in the comments
> but it didnt fix it its doing the same thing 👍
> no infos at the screen but if i press the button near the tip the iron starting
> getting hot like it was suposed to be but nothing on screen :S

Nothing at all, so it's not like in that video, right? Does the vendor
bootloader always handle the screen properly or do you also get random
artifacts there?

lampros2213, 2020-10-07:
some times the screen does some artifacts and with original firmware but if i disconnect and reconnect the power it goes away
can be the cable or the power supply??

NoSkillSammy, 2023-02-15:
I've tried everyone singles person idea forwards and backwards, nothing , absolutely nothing. Tried looking for a cheap replacement board off ebay but seemed like a risk so i just bought a whole new $80 ts100!!!! yay open source firmware :(