Ralim/IronOS #422: Various new bugs with the latest 2.06 firmware on a ts100

sefer2, 2018-12-29:
Hi,

In installed the latest 2.06 firmware onto my ts100. I upgraded from the official 2.18 firmware.
The first thing I noticed is that the main screen flickers twice before settles whenever the solder iron is plugged into power.

Next, entering "Advanced Options" there's a checkbox square to the right of "Detailed idle screen", its right most edge flickers all the time. In fact, this appears to happen with all the items that draw the right-most column on the screen.

Lastly, when I enable the iron, it shows only "305C" (it does heat up, but the screen displays a constant 305C, not showing the current temperature) and no warming/cooling progression. I have no idea when it reached its final designated temperature.

I tried doing a factory reset (I hope I did it correctly) without any effect.

federck, 2018-12-29:
Also I was able to find that:
-The writings in "Tip Model" option are overlapping.
-"Advanced calibration" and "Simple calibration" are not shown in the menu.
-If you set a "boost" temperature near to the maximum limit of temperature and then, during soldering, you activate the "boost mode", the message "bad tip" will be shown.
The question is: would be better reduce the overshoot or maybe reduce maximum temperature depending on the tip model?

Thanks,
Federico

iLLiac4, 2018-12-31:
I have also tested the new firmware and besides the bugs above there is also screen tearing when text is scrolling.
I really like the thermal mass of the tip and would it be possible to implement with the tip model also a small icon of the tip or. is it overkill?
[https://www.google.com/url?sa=i&source=images&cd=&cad=rja&uact=8&ved=2ahUKEwiJ3sa8nsnfAhVH3qQKHTr9AqYQjRx6BAgBEAU&url=%2Furl%3Fsa%3Di%26source%3Dimages%26cd%3D%26ved%3D%26url%3Dhttps%253A%252F%252Fwww.renaissancerobotics.com%252FTS100tips.html%26psig%3DAOvVaw0plRK6XFn0V_aJzxfrF3db%26ust%3D1546316437145929&psig=AOvVaw0plRK6XFn0V_aJzxfrF3db&ust=1546316437145929]

Ralim, 2018-12-31:
[TS100A_2_6_AtollicBuild.zip](https://github.com/Ralim/ts100/files/2717736/TS100A_2_6_AtollicBuild.zip)
Any chance you could give the above build a shot?
Its build using my local machine rather than linux and want to narrow down if this is a compiler issue or not.

@sefer2 I cant reproduce locally the flickering much, can you see if the above build is better or worse?

@federck The text running into each is interesting ,what language are you running? This is most likely a language related issue (I only test EN here usually).
Advanced calibration is only shown when its a user tuned tip, not when you select a preset.
Do you mean the temperature on the screen doesnt show fluctuations or that its not showing live data at all ? When your using it with the new PID the tip temperature fluctuations are reduced considerably.
This new build has a slightly lower cap to try and reduce the ADC bad tip protection, i might re-work that more in a day or two.

@iLLiac4 Basically out of room at the moment to fit any :( (some builds are down to single digit free bytes in the binary). If someone (me or others) impliements compile time font generation we could free up some room for more graphics.

iLLiac4, 2018-12-31:
Have tried it and the flickering seems to be gone.
There is still bad tib if boost mode is selected.
And also new bug that turns off the soldering iron if you do not select detailed soldering info under advanced settings.

federck, 2018-12-31:
Well, I think that the italian release, like other releases using a certain number of characters, it's affected by the problem just because text isn't drawn in the most right part of the screen.
[image]

Yes, I meant live data as @sefer2 did, but that's a problem circumscribed only to graphical interface.
However, in your newest relaese, soldering mode (using GUI) leads to the reboot/freeze of the iron.

Thanks for the other clarifications!

Jaroslaw1, 2019-01-01:
Calibrate temperature still not working. Only i have this problem :) ? When calibrating is done appear: "calibrate tip offset". Temperature showing in menu 005C. Temperature on my room is 23C. Please see video. Thanks and good luck
[image]

https://youtu.be/2yat2SqHzoQ

sefer2, 2019-01-01:
The latest build seems to solve most of these issues I've encountered, with the exception for the double blink when the power is plugged in. I no longer see the flickering with this build.
Great work Ralim! Thank you for a fantastic firmware!

Jaroslaw1, 2019-01-01:
@sefer2 You don't have problem temperature calibration? Maybe I'm doing something wrong?

sefer2, 2019-01-02:
I didn't try that, most of  the original problems I reported appear to be solved though.

Ralim, 2019-01-02:
@Jaroslaw1 Interesting to hear the temp isnt fixed, I'll look at this shortly. Honestly was focusing on other bugs. The temp issue is known about, and its mostly a UI thing, the tip tracks fairly accurately (on my units).

Jaroslaw1, 2019-01-02:
Hey,

Before calibration temperature in main menu is ~15°C, 8°C less than the temperature in the room. After calibration is ~003°C. When i begin soldering, temperature 160°C it melts the tin [solder]. In ver-2.05.1 temp. after calibration is correct. Otherwise everything is fine [best firmware ever!].

Thanks for your work and Regards

iLLiac4, 2019-01-02:
I have tested calibration and it really does not work. When calibrating it sets to around 0 when at room temperature.

federck, 2019-01-02:
"Simple calibration" and "Advanced calibration" show "Detailed Idle screen" description.

mixxxlalex, 2019-01-04:
Hello. I tried the version v2.06-RC2. In it, as in version v2.05.01, Russian letters are displayed incorrectly. Is there any way to fix this?

RustyWraith, 2019-01-09:
> @sefer2 You don't have problem temperature calibration? Maybe I'm doing something wrong?

I can confirm the problem on ts80. Room temp is 0 degrees after a calibration.

ElenaPower, 2019-01-10:
I can confirm 0°C at room temperature after calibration on my TS100.

Cherenkov11, 2019-01-10:
Yes, same bug, after you do the calibration shows ~0 C, and for me I checked the temperature at 300 set, and the external thermocouple shows around 325 C, so a little bit of overshoot. (B2 tip)

csdpm, 2019-01-11:
V2.06 RC2:
- Yes, I see the ~0'C tip temperature at ambient.
- L/R swapped on display orientation
- Cooldown blink not working (simple soldering screen, detailed idle screen)
- Boost and Standby temps appear to track the operating tip temp as it is adjusted up and down.

Apart from that, it's a great usable release.

iLLiac4, 2019-01-12:
Today I have got a new TS100 with PCB v2 and the rotation of the screen for left and right handed is acting strange. It tilst too fast in right hand position (i am left handed). Can this be tweaked?
On the other TS100 with PCB v1 it works as it should.

bakerenko, 2019-01-23:
No russian language, please fix.

Brassee, 2019-01-25:
> V2.06 RC2:
>
> * Yes, I see the ~0'C tip temperature at ambient.
> * L/R swapped on display orientation
> * Cooldown blink not working (simple soldering screen, detailed idle screen)
> * Boost and Standby [...]

Same here

Nazosan, 2019-01-28:
So I just tried this update from a few hours ago on my TS80 and now it won't even turn on.  The only function that works is flash mode by holding + while plugging it in.

iLLiac4, 2019-01-28:
Same on TS100. Does not power on after RC3. Only DFU works so have downgraded to latest available online.

bradanlane, 2019-01-28:
I have a TS80 which came with the USB-C cable and a 9V 2A power supply. The TS80 shows 18W on the status screen. When I use a 5V 2.1A source ithe TS80 only shows between 5W and 5.3W. I’ve trated several wall chargers and USB batteries with a 2.1A output.

Nazosan, 2019-01-28:
The TS80 isn't supposed to really use 5V at all except for the USB connection.  I think also 5V 2.1A chargers expect the device to do something or other to indicate a larger current is safe and fall back to 1A max otherwise (which would be consistent with the results you got.)  Official documentation and specifications explicitly state 9V 2A via QC 3.0 is the only thing it supports.

BTW, I was getting 3C for my tip temperature when it was at room temperature before too on my TS80 even after calibration.  It didn't seem to be that big of a deal since heat mode seemed accurate enough, but anyway, TS80 was doing it too.

whitehoose, 2019-01-28:
bradanlane
Nazosan is correct, the iron is QC3  Withour negotiation the default is 5v, and mine will sit at the menu screen quite happily on 5v - but as soon as the iron starts working it negotiates more volts and heats up using 8 or 9v.
The stock 1.06 would give a low volts error as soon as a button was pressed if connected to USB2
I first came to Ralim looking for f/w that would allow me 5v working (just in case).  But the bugbear is this is (almost) exclusively a QC device - and to make it worse - it can be very picky about which QC it will and wont work with.  If you buy a power brick ... or even a wallwart do your homework and be sure other people are using that model with the 80 too.  You can't even be certain that even a manufacturer is good ... you need to be model specific.

Xiaomi MI 2 - 10000 mAh Power Bank works (I think the 20000 works too - but you need to ask that question).
Search ebay
QC3.0 QC2.0 USB DC-DC Buck Converter Charging Step Down Module 6-32V 9V 12V J6N1 works but needs a 12v source so it has something to step down.  I've combined it with a step up/down buck/boost converter so any source above 3.6v can be boosted to 12v or more to provide enough to get the Iron working
And the DS.OX wall wart that comes with the Iron's "extras" works.
The ISDT BattGo BG-8S Smart Battery Checker Balancer has a QC3 output - and that works too.

If you look through https://github.com/Ralim/ts100/issues/349 there are plenty of other suggestions that work.

bradanlane, 2019-01-28:
whitehoose - thanks for background and issue reference. I wanted 5V support for emergency/backup. I have plenty of USB2 batteries which work - albeit slowly. If I opt for a new battery, I’ll het one of the USB-C models with proper voltage and wattage.

whitehoose, 2019-01-28:
Didn't make that clear .. the compatibility issues are with the USB-c models (QC3).  The standard USB2 models only sort of work depending on the firmware you are using.  The firmware that came with the iron originally didn't work past the first menu screen on USB2 a quite early version of Ralim's firmware did ...

If (when) you go for USBc - do that research - the QC3 "standard" is anything but standard.

ElenaPower, 2019-01-28:
V2.06-RC3 does not work for me. Only black screen.

Brassee, 2019-01-29:
TS100, EN V2.06-RC3, nothing on screen

asymcon, 2019-01-29:
My TS80 does this with 2.06RC2 using QC3:
[VID_20190129_162500.zip](https://github.com/Ralim/ts100/files/2808346/VID_20190129_162500.zip)
Is this voltage pulsating behavior normal? Also it negotiated for 12V even on 18W setting.

asymcon, 2019-01-29:
> So I just tried this RC3 update from a few hours ago on my TS80 and now it won't even turn on. The only function that works is flash mode by holding + while plugging it in.
> 
> A previous one (I think RC2? EDIT: probably RC1) worked fine. Unfortunately previous TS80 files are gone now so I can't downgrade. I didn't realize I should keep backups.
> 
> EDIT: Side note, I just bought a new power supply. It is much higher quality than the one that shipped with my TS80. It is capable of a very high wattage -- 40 something I think -- though of course still the standard 2 amps at 9V via the normal USB 3 port (can QC actually do 2.66 to get 24W or isn't that PD only? I don't really need 24W, I just wanted something better quality that I could be 100% sure would handle the 9V 2A with room to spare.) I mostly did this because the stock one seemed really cheap and some component smelled like it had gotten too hot to me. Anyway, with the version of this custom firmware I was using before this would not work. While heating up it would shut off and reset every time I tried it. I assumed the power supply was faulty. On a whim I tried it with the stock firmware (which I'm now having to use since the custom firmware doesn't boot and I must say that it's definitely much clunkier to actually do the simplest things) and it works fine. Voltage goes down a tiny bit (8.57V while heating) then stabilizes at 9.03V once completed. At this point I wonder if there is anything at all wrong with the plug after all or if it isn't something specific to the custom firmware. I really don't like the stock firmware, but I'd feel more confident using this presumably much higher quality power supply. Does it just not like the lower voltage? Seems to work fine though. It still heats up very fast and holds steady.

Here, I made a backup (sorry @Ralim ):
[TS80_EN.zip](https://github.com/Ralim/ts100/files/2808401/TS80_EN.zip)

Nazosan, 2019-01-29:
Thanks.  Is that RC2?  I'm wondering about the above user's description of voltage issues and it does say something about voltage oscillation while heating.  I wonder if that might be kind of harmful to the iron.  I'm sort of inclined to stick with the stock one even though I kind of hate it (in particular the fact they didn't switch the buttons in left handed mode drives me nuts.)  The fact it seems to give such consistent voltage and works with what has to be a better quality power supply makes me think I would be better off waiting for another version anyway.

asymcon, 2019-01-29:
> Is that RC2?

Yes

> voltage oscillation while heating

Only on 24W and only on QC3.

FiveOhCS, 2019-01-29:
I don't get any activity at all on 2.06 RC3. No flickering or anything. 

Reverted to 2.05 and it works fine.

RustyWraith, 2019-01-29:
No activity on TS_EN.hex from v2.06 RC3.

I cannot figure out how to compile/flash RC4.
Little help here?

Ralim, 2019-01-30:
@rozwell  @RustyWraith My apologies, hex files uploading now.

Nazosan, 2019-01-30:
This "RC4" build definitely works on my TS80.  Cool tip temperature shows correctly in this one too -- or at least really close (it was a few degrees above actual ambient, but I couldn't give a specific number.  Very close anyway.  Much closer than the 3C it said before.)  It still won't work with the (presumably) better power plug that the stock firmware will though.  Watching the voltage on the screen, it's going up and shuts off right as it hits 8.9V for some reason.

From what I'm understanding after trying to read up on the QC 3.0 "standard" is it seems like there are three voltage profiles.  3-6, 6.2-9, 9.2-12 at 3A, 2A, and 1.5A I think.  Is it possible it does something like trying to go to the next range as it hits 9 but fails or something?  I notice when I use the original charger it oscillates as high as 9.8V but never shuts off.  And it smells worse than ever (I think the one they sent with the TS80 really does not like that oscillation,) so I seriously don't trust it.

Ralim, 2019-01-30:
@Nazosan It sounds like your "nice" plug isnt rated for higher than 9V, can you confirm this?

In the 18W mode the iron measures the tip resistance and calculates the ideal voltage to meet the 1.5A.
What firmware language are you using? Ill make you one hard coded to 9V to test this theory with.

The features that will come in future firmware is to allow you to set 9V/12V or 18W/24W modes for the TS80. Just want to get the current features stable first :)

Nazosan, 2019-01-30:
I'm using the EN firmware.  This QC 3.0 charger can probably do 24W mode if that's somehow better.  I just thought a lower wattage setting would mean the iron is working less hard (I want this thing to last a long time.)

rozwell, 2019-01-30:
Flashed RC4 with TS100_EN.hex and it works - great job @Ralim ! :)
The only issue I can spot at this point is invalid tip (BC2) calibration temperature ~34°C, rather than real room temperature 21~23°C.
To confirm, I flashed to stock v2.18 and it showed ~21°C, then back to RC4 and got the results above. Every time with factory reset, just to be sure.

Nazosan, 2019-01-30:
Maybe a simpler solution might be if there was a manual override on tip calibration?  Is it linear?

elgab, 2019-01-30:
v2.06-RC4 flashed on TS100 PCB v2, and working fine.
1. auto rotation set to manual, auto mode is a bit dodgy but nobody is changing hands so...
2. on B2 stock tip i get 26 celsius after calibration, however the room is 21.
3. sleep working fine: 1 minute with 90 celsius.

Trezona, 2019-01-30:
Just flashed TS100_EN.hex v2.06-RC4, and all is working great now. Thanks @Ralim

iLLiac4, 2019-01-30:
Have tested RC4 and it works really good. On PCB v2 auto rotate for left hand is not as good as on v1.

Ralim, 2019-01-30:
@Nazosan Before you claim it can do 24W, double check it can do that in QC mode, a lot limit you to 1.5A in QC mode (even if PD mode is higher).. If it works at 2A.. Well thats up to you to want to test.

24W mode tells the iron to negotiate for a higher voltage than it calculates for 18W basically, to try and push more power through it. Some QC chargers are rated for 12V 2A, though not many. Most people use this to try and run the iron at 12V and to hope that their PSU copes with it :D

@rozwell When you do the "simple" calibration, its only as accurate as the temperature difference between the tip and the inside handle temp. If you want  to have a more accurate scale you can do the advanced calibration with a tip temperature sensor (such as the FG-100 clones).

The tip is roughly linear at higher temps, but a bit more non linear at room temperature, so most errors show up at low temps.

Also this upgrade reset the system settings, so you may want to leave the iron for an hour or two so that the handle and the tip are equal temps, and then perform the simple cal if your temp is offset, though 2-3 degC is better than a lot of other irons out there (1%)

Also when comparing stock to mine, its a completely different temperature compensation and lookup routine being used, my firmware goes for assuming a worst case cal value (slightly high) so that it works for TS100 and TS80, rather than a specific tuned value. 

If you go into the debug menu (hold the rear (B) button while on the idle screen, you can use the front (A) button to go through readouts, and one of these is Than, which is the temperature in the handle x10. This is the only way the iron can actually know the ambient temperature.

@elgab default of 320C is just a common value that is used a lot, since its trivial for anyone to adjust from that point.

I agree that rotation detection isnt ideal on V2 pcbs (TS80 is always a V2). The ST accelerometer does not give as many nice options for tuning rotation detection.

@vasyapupkinodin This is known as the Russian language files try to use Russian chars in the slots that are "one English char only". At some point this will be fixed to support non English characters in these slots.

rozwell, 2019-01-30:
@Ralim Thanks for the handle temp advice - tried calibration after whole night and it was off by only ~5°C.
I checked THan and it was at ~25°C, ambient 20°C so that's where the difference is coming from. From there it just gets higher, when handle warms up while it's powered up, or simply from body heat, resulting in a ~10°C+ difference in a very short time, without even working with the iron.
That gave me an idea for a quick workaround where I just heated up the tip for a few seconds, then used auto calibration to compensate the difference. That did the trick, not practical, but works. 
Looks like the handle temp offset is needed. Or is it already there?

@elgab +1 for default 250°C, much safer. But for now you can set it on USB power - it won't heat up because of undervoltage and when you hold button A, you can set desired temp.

Ralim, 2019-01-30:
@rozwell The thing is that we dont actually care what "ambient is" we only use the handle temperature for cold junction compensation of the thermocouple in the tip. 

The only time handle temp is used for anything other than cold junction calibration is when you do a calibration of the tip, which is why i recommend leaving the unit alone for an hour or two to let its temps stabilise. 

Changing the default setpoint is sort of a moot point since everyone is going to have their own preference, and for those few who dont change it, it wont be the deciding factor in tip damage usually. Tips are an expendable item eventually, so the choice is always yours about changing the temp.

rozwell, 2019-01-31:
@Ralim understood, seems just my unit shows a few degrees higher than it should. Not really important and relatively easy to compensate with the calibration trick I mentioned before.

Looks like it's time for me to finally start using TS100 with your awesome firmware :D

Nazosan, 2019-02-13:
Did you ever do the hardcoded voltage firmware?  I'm curious to try it.  I really do not like the stock firmware, but I'm just afraid to use the power plug they provided with it.

Ralim, 2019-02-13:
24W mode is 12V hard coded, next rev will have more options when i have some time to sit down and write it

Nazosan, 2019-02-13:
Oh, I'm sorry.  I completely misunderstood what you were saying before.  Hmm.  It has an interesting result with 24W mode.  It cuts off as the voltage is slowly rising when it hits ~9V or so, then if I try using it again it works without a problem, this time going up to 12V (where it seems to hold very steady at 12.0V with no oscillation btw.)  I understand it can accept higher voltages than 9V, but is 12V a strain on the TS80 long term?  The wattage rating says 29.7W while it's heating up though.  That's ind of strange.  Officially the type-a port (QC 3.0) only supports 1.5A at 12V.  The type-c (PD) port can do 2.3A at 12V, but that's 27.6W.  Is the TS80 reading it wrong?  The plug doesn't seem at all warm (compared to the plug that came with it getting quite hot even at 18W) but it seems like I smell something from it too.  Is that just a normal result from pushing such a high current through a power source like this?  (I'm curious if the QC 3.0 chargers that came with other people's TS80s were doing this.)

Nazosan, 2019-03-05:
So far the 24W mode seems to be ok, but sometimes for some reason it loses the higher voltage and drops to 5V (I think after sleeping for a while, but I'm not sure if that's the cause.)  It's trying to keep the tip temperature up, but ends up only able to maintain around 250C or so with only 5V.  I have to disconnect and reconnect it to get it back over 5V.

bradanlane, 2019-03-05:
I’ve seen the “stuck at 5V” issue with the 2.04 release on my TS80 and a standard 18W setting. I had the same workaround.

Ralim, 2019-03-06:
The stuck at 5V issue usually means that the QC negotiation failed. I've had this when a plug wasn't seated properly as the initial negotiation is only done once at start as its fairly slow.

Nazosan, 2019-03-06:
Yeah, I assumed that it was such a thing since 5V is presumably the default fallback, but why would it drop to 5V after that initial negotiation?  Does sleep mode actually go ahead and drop back to 5V then renegotiate?  Because if sleep mode continues at the full voltage and merely uses less current then presumably the voltage wouldn't change.  I was thinking maybe it was the plug itself, but I don't dare try running 24W mode to test on the original plug since it appears to be of exceptionally low quality and I'm still afraid to trust it for any length of time.

I did try switching back to 18W and switched back to the original plug, but oddly enough it seemed to keep turning off the heat.  I don't know if it was just resetting or simply disabling heat because it always happened while I was actually using the iron and I didn't know it until after I started having a lot of troubles with the solder and then looked at it.  I think maybe for now I'm going to give in and go back to the stock firmware even though it's really horrible to use (especially if you're left-handed since left-handed mode only switches the screen but not the buttons...)  I don't think the voltage dropped to 5V, but perhaps it did and that would explain it sort of "shutting off" there.  (It's worth noting that the higher quality plug I've been using does support a pretty significant amount of current at 5V.)

I believe the cable was well seated on both sides the entire time.

omair94, 2019-03-11:
I have noticed that when using the TS-80 with a QC 3.0 power bank, there is a faint buzzing noise coming from the power bank. I have tested it with 5 different power bank, all of which exhibit this. When using the included wall adapter, there is no buzzing. The buzzing only occurs on this firmware and not on the official firmware.

I'm not sure if this is actually an issue, but it is a bit concerning.

VjyedSC0UFMc8z, 2019-03-28:
This was so helpful. You have the only firmware that I could get to work with my QC3 power bank https://www.ravpower.com/download/ravpower-turbo-20100mah-power-bank-with-usb-c-and-qc-3.0.html 
All the rest wouldn't trigger the bank to provide power and heat up. 
The only critique I would have for this combo is that my power bank will sleep before any of the standby settings on the TS80. But it turns on, which is a huge improvement. 
Thank you!

whitehoose, 2019-03-28:
This was covered in some detail in the Mamoth ts80 thread (this is for ts100 which isn't really a qc3 candidate)  The QC3 "standard" has been reverse engineered and thus (literally) subjected to the chinese whispers syndrome.  It's always worth asking for powerbanks or adapters that are known to work (have a look here https://github.com/Ralim/ts100/issues/349 - it's been covered a few times)
The falling asleep issue has also been covered, to date the "best" solution (assuming you've tried any inbuilt measures the bank may have) is to either plug in a LED torch/lamp dongle or, if you're any good with a soldering iron o<}:oP  stick a resistor and LED on the end of a cut down usb lead yourself.  The additional constant load will do the trick.

asymcon, 2019-03-28:
Xiaomi Mi Pro powerbank has a special mode for charging low power devices which instead of its typical 30 second timeout, turns off after 4 hours.

[16 more replies]

closed 2019-12-29 by Ralim