Ralim/IronOS #429: TS80 temperature indicator is still flickering

RustyWraith, 2019-01-05:
* **I'm submitting a ...**
  - [x ] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Report a bug

* **What is the current behavior?**
Tip is heating and cooling down with smooth +1 degree increment. Reaching setpoint temp indicator starts to jerk +/-50 degrees. Heating indicator never shows "heating" picture.

* **What is the expected behavior?**
Stable +/-1 readings.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
Heat an iron to setpoint.

***Steps to reproduce:***

1. Heat
2. Wait
3. 

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

  - Firmware Version: V2.06, 1.06 stock, 1.07 stock
  - PCB Version: TS80
  - Power Supply (Voltage and Current Rating) : stock 9V/2A, led 12V PSU, 12V ATX PSU

* **Other information** 

I've tried different PSUs and cables after reading #261.

epigramx, 2019-01-16:
If you mean the temperature never stabilizes, I got the same today on a ts100. I returned to the official one that easily goes to a 300 and stays there.
PS. If it helps, the official one at first did have the temperature jump around (at first start) but the it stabilized.

Ralim, 2019-01-16:
Remember that the stock firmware still does this, it just hides it from you :grimacing: 
There is work to be done on the TS80 tune that will help with this. Current tune on the ts100 is very very good, but I haven't been home to do a tune on the ts80 recently.

dhiltonp, 2019-01-17:
@epigramx: what temperature range fluctuation are you seeing on the ts100?

RustyWraith, 2019-01-18:
@Ralim 
I know, that some fluctuations are inevitable.
I am confused with an amplitude and the fact, that the stock firmware makes it too.

dhiltonp, 2019-01-18:
I've had wide fluctuations on a damaged tip.

RustyWraith, 2019-01-18:
How can I tell? It should be brand new.

dhiltonp, 2019-01-18:
With Ralim's firmware, I get fluctuations of 1C around the target temp. On my damaged tip, the temp may fluctuate 10C around the target temp when idle!

The tip still works, but precise temperature control is difficult.

epigramx, 2019-01-18:
> @epigramx: what temperature range fluctuation are you seeing on the ts100?

only 1-2C at the end of the day; not a big deal
(I also have a BC2 (I tested a B2) but I haven't tested that yet)

RustyWraith, 2019-01-30:
Temperature jumps on RC4.
https://youtu.be/KijwWwjP8qI
Will try to order a new tip to exclude tip malfunction.

Ralim, 2019-01-30:
That sort of jumping is not something that I have seen on any of my units :(
I think that is similar to what others have seen with a damaged thermocouple.

dhiltonp, 2019-01-31:
There is definitely something unusual about the thermocouple or its thermal connections on that tip.

RustyWraith, 2019-03-02:
Yep. A new tip works perfectly.

discip, 2022-04-23:
@RustyWraith
Since this is a hardware issue this should be closed. 😊

thanks

closed 2022-04-23 by Ralim