Ralim/IronOS #431: TS100 Current protection - Feature

iLLiac4, 2019-01-14:
* **I'm submitting a ...**
  - [ ] Bug report
  - [ *] Feature request
  - [ ] Translation

In case if you are using lets say 24v 9A power supply and if the tip fails, then TS100 will use 9A and this will in best case scenario destroy mosfet in TS100 iron.

Is it possible to limit the iron to use max lets say 4A and if exceeded it will shut down automatically?

whitehoose, 2019-01-14:
Quickest, easiest would be to fuse the output but easiest isn't always best .
It depends just what your aim is.  Is it solely to preserve the iron or as a safety measure to stop the batteries going Chernobyl on you as well as preserving the iron? 
Car fuses will accommodate the current... but it wont be a particularly fast blow and still needs the cell in a firebag.
there are lots of protection circuits out there designed primarily for LiLon but unless it says, OK with LiPo too
This is a little frightening, but covers all the possibilities!!! 
https://www.youtube.com/watch?v=w8xtSTspFCE

I haven't used this - but it shows the connections
https://www.amazon.com/UEB-Protection-Li-ion-Lithium-Battery/dp/B07DQPFX5N

I use a s4 connected through a up down buck boost and QC3 in my ts80 pack and switch modules in and out depending on what I'm powering.  It's not pretty but it seems to work and should capture most "blips". 
It usually easy enough to intercept the outputs at the balance plug to build in a protection stage between source and load without disassembling the package, as long as you're not using silly currents for extended periods.

Ralim, 2019-01-26:
Honestly there is almost nothing that can be done in software (we have no means to measure current, or respond quick enough to prevent the mosfet detonating). Your only options are going to be physical protections here.

closed 2019-08-27 by Ralim