Ralim/IronOS #433: Auto-off display at main (solder) mode

Pilson66, 2019-01-17:
* **I'm submitting a ...**
  - Feature request

* **What is the current behavior?**
OLED display is alway on in main mode. 

* **What is the expected behavior?**
OLED display is auto-off after constant (or custom, selected from menu) time period, if no buttons was pressed at this period of time. 

* **What is the motivation / use case for changing the behavior?**
OLED displays "burn in" over time, so this feature would significantly extend its running time.

* **Other information** 
Also, it would be nice to have this feature  off/on in the menu.

Ralim, 2019-01-17:
Hi,
This already exists in the firmware and should trigger after about 5 minutes off memory...

Pilson66, 2019-01-17:
Yes, but no. 
I installed the firmware version 2.0.6-RC2, and used the soldering iron for 20 minutes. During this time, the display has not turned off. I emphasize that we are talking about auto-off the display in the active mode of using a soldering iron, not in lying in place.

Ralim, 2019-01-18:
Ah okay, I get you.

I wouldn't advise to have the display sleep during use because the amount of time you are soldering with the iron is so small compared to the lifetime of the OLED display. (Probably 5-10 years at 8 hours per day, every day).

Having the screen on is more convenient for comparison than having to push a button to wake the screen.

closed 2019-08-27 by Ralim