Ralim/IronOS #434: Accelerometer tweak for left hand user

iLLiac4, 2019-01-19:
I am using latest firmware 2.06rc2 and have two TS100 irons one is based on PCB1 and second on PCB2.
I am left handed and I remember that you tweak the autorotate feature some time ago for version PCB1 that it works great also for left handed users.

Can you do please the same for PCB2 version since it rotates the display too quick into righ hand position.

No changes are needed for PCB1 since it works great. If the PCB2 would work the same as PCB1 it would be great.

TNX

discip, 2022-04-23:
@iLLiac4
Have you tested the latest release yet?
If that solves your issue, please don't forget to close it. 😊

thanks

River-Mochi, 2023-03-24:
Some of these  older tickets are more than 2 years old and don't relate to any  ongoing or future firmware fix.
  perhaps a notice could be put on them  asking the original poster to respond  within  45 days and if not,  the ticket is closed?

closed 2023-03-24 by discip