Ralim/IronOS #436: Temperature calibration and tip detection

RFC3251, 2019-01-25:
I've seen a couple of other people mention "temperature shows as zero", but my symptoms are slightly different, and they were using a different firmware version, so here goes:

Note: the following was done in a room at approximately 17 degrees, the tip is a BC2 (and is set to BC2 in the menu) and I'm using FW 2.06-RC2. 

**Issue #1: Tip temperature shows only as an offset (i.e., zero when at room temperature)**

I left the iron alone for several minutes (not touching the handle or the tip), and then (still without touching the handle at all), went into advanced -> calibrate temperature -> (press tip button again).

At this point, the instructions on the main page say "The display will go to "...." for a short period of time as the unit measures the tip temperature and the handle temperature and compares them The display will then go back to TMP CAL". Mine did not show TMP CAL, it just showed the scrolling text after the three dots. I assume "TMP CAL" was from older versions, and this means it finished calibrating? Would it ever display an error message (like the OEM firmware)? Some feedback like "OK" or "Done" would be nice.

Anyway, after this was done (still without touching the handle, so it doesn't heat up), I scrolled back to the (detailed) idle screen, and it showed the tip temperature oscillating between 0 and 1 degrees. Is the idle screen supposed to show just the _difference_ between the tip and the handle, or does the handle sensor return an actual temperature? If so, it looks like the FW is "forgetting" to add the handle (ambient) temperature to the value being displayed.

**Issue #2 (probably related): Disconnected tip shows "400" while cold tip (with warm handle) shows "DISCONNECTED"**

Also, after I held the handle in my hand for a while (meaning the handle was now slightly _warmer_ than the tip), it stabilised at 0, and then started showing "TIP DISCONNECTED" (although a tip was inserted). When I actually removed the tip, it instead showed "TIP: 400". So it looks like the logic being used to detect if a tip is inserted (is it based just on temperature?) isn't working properly, either.

I tested this both when powered through the DC jack (22V, from a lab PSU) and through USB, from a monitor hub. There was no difference in behaviour.

RFC3251, 2019-01-30:
RC3 has fixed the temperature offset, but removing the tip still shows a high temperature (now "431 ºC", at an ambient temperature of about 20 ºC) instead of "TIP DISCONNECTED".

Ralim, 2019-01-30:
Yeah, its getting harder to be able to pick between a "valid" and invalid reading now, especially on the TS80 which tops out at around 430-440 depending on ambient and tip gain.

I need to put more time into a better solution for future.

RFC3251, 2019-01-31:
So the only way to detect if a tip is inserted is by looking at the temperature? There isn't a built-in way to just check for continuity between the tip contacts?

I haven't really looked at the hardware, but I believe the iron switches between heating and reading temperature, right? Wouldn't it be possible to check if any current is actually flowing during the "heating" phase?

Ralim, 2019-01-31:
@RFC3251 There is no way to know "If current is flowing". So the only way to know is to look at the temp reading and check for if it is reading maximum (ie the pullup has pulled it high).

There is no fancy way to check continuity or anything fancy.

Honestly tempted to remove the whole tip removal detection stuff and let people use their eyes.

RFC3251, 2019-01-31:
Well, I wouldn't call hardware tip detection "fancy", but I guess Miniware didn't think it was a priority. And I guess tips don't necessarily fail open, so maybe the situation in which a tip is inserted but "not connected" would be pretty rare?

So yeah, if the only way to detect a tip is to look for "unlikely" temperatures, I guess it's probably not worth taking up FW space with that, since it might a) not catch real errors and b) report false errors.

closed 2019-12-29 by Ralim