Ralim/IronOS #437: python converted logos, not flashable

noemu, 2019-01-25:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **What is the current behavior?**
with python created boot-logo result in an "err" file, after flashing.
I compiled the gui-logo-converter and this program creates a perfect flashable file. 

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. ```python img2ts100.py in.png out.hex```
2. drag and drop hex on the ts100 in dfu mode
3. receive err file

* **What are you running:**

  - Firmware Version: 2.05
  - PCB Version: 2

* **Other information** 

[lgos.zip](https://github.com/Ralim/ts100/files/2797129/lgos.zip)

Ralim, 2019-01-26:
Hi, can you upload the files made by the python code and the c# tool for comparison? Both appear to work for me here, so wondering if its to do with your computers setup?

noemu, 2019-01-26:
[csharp.zip](https://github.com/Ralim/ts100/files/2798867/csharp.zip)

m-kozlowski, 2019-03-14:
Had similar problem. changing line terminators to CRLF solved issue for me.

closed 2019-08-27 by Ralim