Ralim/IronOS #438: Readability and usability

noemu, 2019-01-25:
* **I'm submitting a ...**
  - [ ] Bug report
  - [x] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
I suggest to put a little space between a number and a unit. Or make the unit smaller in the soldering mode.
Second suggestion flip - and + in the temperature selection

* **What is the current behavior?**
  - ```100C```
  - ```+ 360 -```

* **What is the expected behavior?**
e.g.
  - ```100 C```
  - ```- 360 +```

* **What is the motivation / use case for changing the behavior?**

Like described in gestalt laws, the brain automatically tries to complete a C int a 0. And that makes it more difficult to read the actual temperature, if you only look for some milliseconds on the display. Making the C smaller or a little bit separated from the numbers, increases the readability.

The motivation for the second suggestion is to make it more intuitive to change the temperature. a number line goes from left to right. So the Brain associate the left side with low things and the right side with high things. So lowering the temperature would be more intuitive on a button on the left side.

* **What are you running:**

  - Firmware Version: 2.05
  - PCB Version: (2)

RFC3251, 2019-01-26:
"100 ºC"  would be even better, if possible. And while I sort of agree about the + and -, I do think it makes sense for the button closest to the tip to be the "boost" button, and reversing the signs would mean you'd boost by pressing "-".

Ralim, 2019-01-26:
If you can design up nice looking "ºC" and "ºF" symbols that are 16 pixels high and 12 pixels wide, I'm more than happy to change to this :)

However, having the + symbol at the front will probably stay, after every time I've tried the reverse or auto changing people come along and complain loudly :)

noemu, 2019-01-26:
what about the symbols you are already using in the temperature setting?

- main.cpp:383
```
		if (systemSettings.temperatureInF)
			OLED::drawSymbol(0);
		else
			OLED::drawSymbol(1);
```

federck, 2019-01-26:
Just in case, here they are.
Editable as many time as needed. [image] [image]

haarp, 2019-07-25:
I too wish that the firmware would use the correct unit for temperature instead of  `C`.

> If you can design up nice looking "ºC" and "ºF" symbols that are 16 pixels high and 12 pixels wide, I'm more than happy to change to this :)

When you're in soldering mode and adjust the temperature, it already shows a decent looking `°C` character. I'm a bit confused why this symbol isn't used on the main soldering screen aswell.

Ralim, 2019-07-25:
@haarp 
Your welcome to create a pull request if you want this changed faster :) Otherwise I'll try to get to it next time I'm working on this

PR #538 (2019-12-29): Fix temp symbol
Please try and fill out this template where possible, not all fields are required and can be removed.

* **Please check if the PR fulfills these requirements**
- [X] The commit message makes sense
- [X] The changes have been tested locally
- [ ] Are there any breaking changes

* **What kind of change does this PR introduce?**
Tweak to use nicer temp symbols in soldering mode

* **What is the current behavior?**
Fixes #438
closed 2019-12-29 by Ralim