Ralim/IronOS #440: TS100_FR.Hex from 2.06 RC1, TS/100 does not work

jpis, 2019-01-28:
A) I tried to install from TS100_FR.hex from TS100 & TS80 with bug fixes part 2, the RDY file appeared, but my TS/100 was not working afterward (black display)!

B) I reinstalled TS100_FR from 2.06 RC1, TS/100 was working again.
C) I then retried to install TS100_FR.hex from TS100 & TS80 with bug fixes part 2 with same bad result as A)
D) same as B)

It seems that we need another realease :-)

Trezona, 2019-01-28:
Yah, i have the same problem on my TS100 using the TS100_EN.hex file.

rozwell, 2019-01-28:
I just got my TS100, yay! :D
Same here with TS100_EN.hex but I believe you should change the title, cause it's v2.06-RC3

vasyapupkinodin, 2019-01-28:
Hello. I have ts-100 dfu: 3.45 v2.18. On version v2.06 RC3 I have a black screen. On the rest up to v2.04 psb2 I have no text. And only on version 2.04 psb2 works for me. I am from Russia and I am writing through a translator, I apologize for any inaccuracies. If possible, I will be happy to assist, my mail vasyapupkinodin@gmail.com

Bosek, 2019-01-29:
I've just uploaded TS100_EN.hex v2.06-RC3 and I have a black screen. How do I reload the firmware? When I connect it to the PC it just says it's malfunctioning device and won't appear.

rozwell, 2019-01-29:
> I've just uploaded TS100_EN.hex v2.06-RC3 and I have a black screen. How do I reload the firmware? When I connect it to the PC it just says it's malfunctioning device and won't appear.

@Bosek Press and hold A button, connect iron to the PC, then release button A

rozwell, 2019-01-29:
Looks like this is a bigger issue - more info in #421 which makes this one a duplicate.

sefer2, 2019-01-29:
Same issue here. The screen doesn't light up anymore after the 2.06RC3 is installed.

Ralim, 2019-01-29:
Please keep discussion to #422 as per the comments on the release page.
Closing this issue noise.
New release up with rebuilds.

closed 2019-01-29 by Ralim