Ralim/IronOS #446: Feature: Custom GIF on boot-up

Jaqb2137, 2019-02-03:
Feature request

Hi,
I have a simple ( I hope :D ) idea, that besides custom logo on boot-up, You can also implement a custom gif on iron startup. That could be a fun little feature.
Chears!

Ralim, 2019-02-04:
You are welcome to create a pull request for this :) 
It would be rather limited though, total size of all frames would be 1kb, so about 5 frames

Ralim, 2019-12-29:
if anyone would like to work on this,
It would require extending the python script to unpack the extra frames into the remainder of the flash page (aka just join them all end to end), and would need to store a framerate somewhere too.

First job is to decode the GIF into images
Then add these to the HEX.

discip, 2022-01-13:
@Ralim
Is that https://github.com/Ralim/IronOS/pull/1180 sufficient? 😅

kind regards

PR #1232 (2022-03-13): Logo rework

* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [ ] There are no breaking changes

* **What kind of change does this PR introduce?**

Support for new logo support that (a) fixes the old endian issue and (b) allows animation.
Removes logo's from this repo and instead places them in the Meta repo.

* **What is the current behavior?**

* **What is the new behavior (if this is a feature change)?**

* **Other information**:

Closes #1173
Closes #1228
Closes #617
Closes #446
Inches closer to supporting #292
closed 2022-03-13 by Ralim