Ralim/IronOS #447: High Voltage Error

IOIOIOIOIOI, 2019-02-03:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [X] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Report a bug

* **What is the current behavior?**
High-voltage error with the power supply the iron comes with. Used a bench power supply and found the error to occur @ 7.8V. The iron only draws 5W of power and warms up extremely slowly. I believe the temperature reading is also wrong because it struggles to flow solder wire @300degC. It reports a voltage of 3-5V higher than the supply voltage. At times the iron resets and displays "TFU: 3.45".

* **What is the expected behavior?**
Normal functionality

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
Normal use

***Steps to reproduce:***

1. Normal use
2. 
3. 

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 1.06
  - PCB Version: 4c155 MT8RI
  - Power Supply (Voltage and Current Rating) : 6-6.5V 3A / 6.5-9V 2A / 9-12V 1.5A

* **Other information** 

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

whitehoose, 2019-02-03:
@IOIOIOIOIOI 
Hi.  Looking at this I'm a little confused as to what we're talking about here.  Let me say I'm just being a little nosy until Ralim or one of the other leaders picks up.
1 is it a ts80 iron? - that would likely have the 1.06 fw.
2 If yes, is that stock PSU the DN.OX ?
3 The 1.06 fw is a stock firmware - ralim and the others don't have control over minidso's firmware.  Have you tried re-flashing your iron with the appropriate Fw created by Ralim?  Or updating to newer version of stock?
To update the fw restart the iron while pressing button A
1. download the suitable TS80 firmware to your PC.
2. Hold TS80's “A” key, and connect TS80 to your PC with USB Type-C cable, to enter
 DFU mode. A display of “DFU3.45” will appear on the screen;
3. Using file manager Copy the .hex firmware to the root directory of that disk.  You will probably get an error the first time - repeat the process - When the extension of the
 firmware changes from “.hex” to “.rdy”, disconnect USB and the firmware is upgraded.
Reconnect to PSU
Stock MiniDSO http://www.minidso.com/forum.php?mod=viewthread&tid=3208
currently 1.07.
Ralims F/w for both ts80 and ts100 : https://github.com/Ralim/ts100/releases

If TS80 - the device is designed to use QC3 power sources
So - Have you tried any other QC3 enabled power sources??? wall wart, powerbanks etc?

Ralim, 2019-02-04:
@whitehoose Looks to be on the money here, I cant really add much more here. The firmware number you are using looks like a miniware firmware rather than my own. I dont support their firmware here, for support with their firmware please use their forum as linked above.

You can also get really strange behavior like this if you try and flash TS100 firmware to a TS80 by accident.

IOIOIOIOIOI, 2019-02-04:
Good day, thank you for your response.

Yes, it is a TS80 iron with the stock fw. I had these issue out of the box (I haven't reflashed it yet).
Yes, it is the stock DN.OX power supply. I also used a bench power supply set to QC3 values, but the iron reports higher voltages and only draws 5W of power on heat up. 
My apologies for posting on this thread about the stock fw. I am unfamiliar with the different ones that are available.

I will try the MiniDSO 1.07 and Ralim's TS80 fw and see if it fixes the issues.

Thank you for the clear instructions and help. It is much appreciated.

IOIOIOIOIOI, 2019-02-04:
Good day,

I tried MiniDSO 1.07 and Ralim's TS80 fw, but the issue persisted.
I believe I might have a faulty iron, since the reported input voltage is still up to 6V higher than the actual input voltage on my bench power supply. At room temperature, the iron also reports 80degC.

I tried to use the calibrate temp function on Ramlin's firmware, but it still reports 80degC at room temperature.

whitehoose, 2019-02-04:
When you say bench PSU set to QC3 .... I'm not certain just what that means.
There are a some people who are circumventing QC3 ... but to be pedantic about it - the iron is designed to talk to QC to get whatever voltages it uses.  Trouble is the QC3 "standard" is anything but.  It appears its been reverse engineered, it really doesn't know if it's coming or going.
I trust the DN.OX, so that's probably the safest starting point.  Its nice to have another option simply because even a cert comes last occasionally.

I use a Ruideng USB um34c usb tester - and it's all reasonably consistent.
From cold
Vout 5v (whatever the source) begins ID negotiation within 2 sec.
press button A - Iron starts heating QC activates - Iron shows 7.8v - 5w during heating, tester shows voltage "dancing" between 7 and 9.8v - 300c within 35 seconds with d25 tip.  Volts settle at 7v
This has been consistent as long as I've had the iron over a range of the sources that work for me.

If the DN/OX is being used with bad results It does sound like a faulty Iron (or a faulty psu).

If you bought from banggood I've found them pretty good - but they won't shift without a video of whatever you're talking to them about not doing (or doing) whatever it is.   It sometimes needs a little lateral thinking to get your point across

Ralim, 2019-02-06:
@IOIOIOIOIOI It honestly looks like your unit is faulty in its resistor bridge for reading the incoming dc voltage. I would flash stock and contact where you purchased from. This is not a software quirk and will prevent both mine and the miniware software working if its this far out of whack.

closed 2019-08-27 by Ralim