Ralim/IronOS #449: 5v source on motherboard

SirKittyMcMittens, 2019-02-05:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ ] Bug report
  - [X] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Request a feature

* **What is the current behavior?**
everything is behaving as it should.

* **What is the expected behavior?**
I want to wire in led lights that run off 5v inside my ts100. purely cosmetic. I have the schematic but I have no idea how to read it. 

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. 
2. 
3. 

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**
cosmetics. I have a clear shell, so the idea of having leds light up when turned on catches my attention.

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.06
  - PCB Version: 2(?)
  - Power Supply (Voltage and Current Rating) : DC input: 100-240v~50/60Hz 2.0A output: 19v   2100mA

* **Other information** 

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Ralim, 2019-02-05:
Your best bet is to use the 3.9V rail in the iron, as there is no 5V rail.

This is labelled as 3V9 in the schematic  and you best bet to finding it is to look for the common point between R26,R7,C5. 

Note that you will not be able to pull much current from this point, but should be able to power a couple of leds.

SirKittyMcMittens, 2019-02-05:
I'm sorry, but where exactly can I find the 3.9v rail? I'm trying my best to follow the schematics, but I just can't figure it out.

whitehoose, 2019-02-05:
The schematic is a "map" of the circuitboard - use it to look at the mobo, you'll need a magnifying glass to translate the picture to reality.  Use it as an exercise to prep for the job.  I'd use a couple of 0603 or 0402 smd LEDs and resistors - try it out first with 5mm leds to see if it's bright enough - I'd underdrive the LEDS as hard as possible, its a soldering iron, not a christmas tree - you don't want to starve the tip too far 100 ohm for red (ish) - adjust for other colours.
[image]
There is a loud voice in my ear screaming "don't do it", but its your decision.

SirKittyMcMittens, 2019-02-06:
Got it. these leds are extremely tiny. smaller than grains of rice. I also found the trace on the board. Thank you for all the help! :)

closed 2019-02-06 by SirKittyMcMittens

whitehoose, 2019-02-06:
good luck, small and fiddly - but so is the circuit board, things are pretty tight in there.  The LEDs are small but perfectly formed and put out a surprising amount of light - unfortunately they take the same current as "normal" ones.  Thinking about the  situation I'd possibly use wire straps as standoffs rather than attempting to add directly to the board  wire, led and a resistor in series should be pretty rigid and need less than 5mm.  If you are new to SMD - Kapton tape doesn't melt - you can use it to hold components while you solder them, and then tape the underside when you cram components into tight spaces to prevent shorting.  UV glue is also a good potting resin for LEDs.

post a photo when you've done the deed

SirKittyMcMittens, 2019-02-06:
here's a photo of the finished result. it came out pretty dang niffty! https://imgur.com/531Yemp

I also used tiny film on the under side of the screen to change the screen color. matched pretty well to the pink leds I used :) Thanks again for the help kind sirs!